CloverHackyColor/CloverBootloader #104: Screen Garble at boot

danishblunt, 2020-03-27:
When having settings to UEFI and 1080p on Intel HD 4400 (I assume more non native intel GPU's have this) Graphics clover will garble the screen, it's a known issue for a while. It usually gets fixed by setting the screen resolution to 4:3 768p, use CSM (forcing low res) or when logged in sleep and wake. OC seems to have a workaround as the issue also happens however only for a splitsecond. Maybe get inspiration for the fix/workaround from there? :)

SergeySlice, 2020-04-11:
Never hear such issue. All is fine.
I think the problem is somewhere in tonymac.

danishblunt, 2020-04-11:
> 
> 
> Never hear such issue. All is fine.
> I think the problem is somewhere in tonymac.

Wrong, this is on a clean install, no premade bloated garbage like unibeast etc.

https://www.tonymacx86.com/threads/fix-resolve-boot-screen-garble.175799/
https://www.tonymacx86.com/threads/boot-screen-garble-on-sierra.216143/
https://www.tonymacx86.com/threads/garbled-screen.252786/
https://www.tonymacx86.com/threads/screen-garble-after-apple-logo-yoga-500-high-sierra.261964/
https://www.reddit.com/r/hackintosh/comments/89sp5o/updated_to_high_sierra_from_sierra_garbled_screen/
https://www.insanelymac.com/forum/topic/313060-garbled-screen/
https://www.hackintoshzone.com/topic/9626-intel-hd5500-garbled-login-screen/?ct=1586632074

This is a very known problem. So far the only workarounds are as listed on top. I would suggest, to make a "fix" in clover that would just change the resolution to 4:3 768p at the end of loading the last stuff, maybe add it like hw fixes which can be checked of unchecked, this would act like opencore where the screen gets glitchy for a split second, but would boot fine into MacOS and make the selection of the boot at 1080p+. 

You would make a ton of people happy as they wouldnt be forced to use CSM or low resolution in clover.

SergeySlice, 2020-04-14:
8 apples is not an issue. It happens on real MacBook.
There are several workarounds, kext patches or using additional kexts to to break a screen.
Nothing to do with Clover.

closed 2020-04-14 by SergeySlice

brisamediterranea, 2022-01-25:
Is this all?