CloverHackyColor/CloverBootloader #105: Package install location

jwballan, 2020-03-28:
When I try to update Clover-5093 to Clover-5107, the installer package doesn't show the EFI partition as an install location:
#:                       TYPE NAME                    SIZE       IDENTIFIER
1:                        EFI EFI                     209.7 MB   disk0s1

It does show an EFI partition on a USB Disk:
#:                       TYPE NAME                    SIZE       IDENTIFIER
 1:                 DOS_FAT_32 CLOVER EFI              209.7 MB   disk4s1

Is this a partition type issue?  (Note Clover-5093 now doesn't see disk0s1 either)

vectorsigma72, 2020-03-28:
Never happened that a pkg was able to target an EFI partition

jwballan, 2020-03-29:
Here is the diskutil list:
$ diskutil list disk0
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *2.0 TB     disk0
   1:                        EFI EFI                     209.7 MB   disk0s1
   2:                  Apple_HFS Time Machine            1.3 TB     disk0s2
   3:                 Apple_APFS Container disk2         750.0 GB   disk0s3

Here is the install.log:
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: @(#)PROGRAM:Install  PROJECT:Install-1000
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: @(#)PROGRAM:Installer  PROJECT:Installer-965.1
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Hardware: MacBookPro11,1 @ 2.71 GHz (x 4), 32768 MB RAM
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Running OS Build: Mac OS X 10.15.3 (19D76)
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: USER=jwb
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: COMMAND_MODE=unix2003
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: DISPLAY=/private/tmp/com.apple.launchd.Zax0yDVE1j/org.macosforge.xquartz:0
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: LOGNAME=jwb
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.IDlQjC2nki/Listeners
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: HOME=/Users/jwb
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: SHELL=/bin/ksh
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: TMPDIR=/var/folders/bb/c3yjsvks3mjc1ywzyj4tzh_h0000gn/T/
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: XPC_SERVICE_NAME=com.apple.xpc.launchd.oneshot.0x10000014.Installer
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Env: XPC_FLAGS=0x0
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Clover r5107 EFI bootloader  Installation Log
2020-03-29 08:43:23-04 MacBook-Pro Installer[1920]: Opened from: /Users/jwb/Downloads/CLOVER/Clover_r5107.pkg
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Package Authoring Error: <background> has an unsupported MIME type: image/data
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Package Authoring Error: <background-darkAqua> has an unsupported MIME type: image/data
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: Product archive /Users/jwb/Downloads/CLOVER/Clover_r5107.pkg trustLevel=100
2020-03-29 08:43:24-04 MacBook-Pro Installer[1920]: PackageKit: Bypassing the enforcement checker since the destination is read/write
2020-03-29 08:43:25-04 MacBook-Pro Installer[1920]: External component packages (66) trustLevel=100 (trust evaluation failed: Error Domain=PKInstallErrorDomain Code=101 "The package “Clover_r5107.pkg” is not signed." UserInfo={NSURL=#Post.pkg -- file:///Users/jwb/Downloads/CLOVER/Clover_r5107.pkg#Distribution, PKInstallPackageIdentifier=org.clover.post, NSLocalizedDescription=The package “Clover_r5107.pkg” is not signed.})
2020-03-29 08:43:25-04 MacBook-Pro Installer[1920]: Could not load resource license: (null)
2020-03-29 08:43:27-04 MacBook-Pro Installer[1920]: JS: checkFileSystemIsWritable diskutil result = 1
2020-03-29 08:43:27-04 MacBook-Pro Installer[1920]: JS: checkFileSystemIsWritable: file system is writable: Passed
2020-03-29 08:43:27-04 MacBook-Pro Installer[1920]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.
2020-03-29 08:43:27-04 MacBook-Pro Installer[1920]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
[... cut ...]
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
2020-03-29 08:43:32-04 MacBook-Pro Installer[1920]: IFJS: *** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')
$ 
I have install window shots, but they seem to large.

vectorsigma72, 2020-03-30:
yep but this way I cannot see errors as "*** exception: TypeError: null is not an object (evaluating 'my.target.mountpoint')" is not an error until you don't select a volume. What about upload a zipped text file?

danimaldtruth, 2020-03-30:
UNSUBSCRIBE

vectorsigma72, 2020-04-22:
Can't determine the issue without a log file

closed 2020-04-22 by vectorsigma72