CloverHackyColor/CloverBootloader #106: 5107 scan entries reboot.....

suedeyang, 2020-03-28:
Everything worked fine when I use clover 2.4k.
BUT!!
since clover update to 2.5k
every time I boot ,"scan entries" message  shows and then just reboot
over and over again
Please do me a favor tell me what to do
Thanks

[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4397407/debug.log)

vectorsigma72, 2020-03-28:
Try to set GUI->Scan->Legacy = false

SergeySlice, 2020-03-29:
There is no issue with Clover 5107 on Z170 computer.

suedeyang, 2020-03-29:
> There is no issue with Clover 5107 on Z170 computer.
I don't have any clue about it.
same PC,same BIOS setting,nothing change,for Clover 2.4k OK,but not for 2.5k
every time boot to "scan entries....." and then reboot,
I check debug.log,the last line is "14:306  0:018  found AppleSMC protocol"
Is it something to do with AppleSMC?

suedeyang, 2020-03-29:
> Try to set GUI->Scan->Legacy = false

not working...
thanks anyway

suedeyang, 2020-04-02:
> Try to set GUI->Scan->Legacy = false
I disable it and doesn't work

Here is the LOG that I use clover 2.4k 
It Work!
But I can't tell what's going on
CAN SOMEBODY HELP ME!!

[debug2.4K.log](https://github.com/CloverHackyColor/CloverBootloader/files/4419439/debug2.4K.log)

suedeyang, 2020-04-09:
> There is no issue with Clover 5107 on Z170 computer.
I realised where goes wrong eventually.
the keypoint is  "share graphic RAM"
in the begining my share  graphic ram is the heightest 1024M
clover2.4k works fine in the past
But clover 2.5k crash every single time
then I switch it down to 512M
Whole system running smoth as silk.

SergeySlice, 2020-04-13:
Do you know driver folder difference between 2.4 and 2.5?

SergeySlice, 2020-04-19:
10 days passed. The issue abandoned.

closed 2020-04-19 by SergeySlice