CloverHackyColor/CloverBootloader #107: DisabledAML keys not taken into consideration when selecting alternative config

mackonsti, 2020-03-29:
Hello everyone, thank you again for your continuing support for Clover!

I have detected this issue in the last builds of Clover and decided to report it today.
Using Clover **r5107** today with multiple configs saved in the CLOVER folder (of EFI) together with the principal `config.plist` file.

In this folder I have files such as:

```
config-0-sos-installer.plist
config-2-drop-tables.plist
config-4-final-smbios.plist
config-6-mojave-working.plist
config-8-mojave-optimal.plist
config-1-basic-boot.plist
config-3-drop-regions-gfx.plist
config-5-fine-tune.plist
config-7-mojave-sos.plist
config-9-mojave-devices.plist
config.plist
```

If I select anything else in the main Clover boot menu, except default `config.plist` then the section _DisabledAML_ section and its keys are not taken into consideration, seems ignored (I confirm this when I run IORegistryExplorer and check for injected values). I am forced to copy that selected plist as `config.plist` in order to have these AML files ignored (inside ACPI/patched folder of EFI).

Is this expected behaviour or a bug?

Here is a sample of my DisabledAML entries in e.g. `config-9-mojave-devices.plist` that I wanted to experiment with, without touching `config.plist` for example:

```
<key>DisabledAML</key>
<array>
	<string>SSDT-GIGE.aml</string>
	<string>SSDT-HDEF.aml</string>
	<string>SSDT-SATA.aml</string>
	<string>SSDT-UIAC.aml</string>
	<string>SSDT-USB.aml</string>
</array>
```

Thank you in advance for your feedback. Let me know if you require further information on my part, happy to provide it.

mackonsti, 2020-03-31:
@vectorsigma72 and @SergeySlice can I bother you to quickly look at the Clover code and confirm if alternative `config` files ignore DisabledAML files by accident? Thank you again

SergeySlice, 2020-04-11:
I didn't check in the real work but as I see DisabledAML feature should switch reading new config.plist.

mackonsti, 2020-04-11:
Thanks @SergeySlice but I have tried some times and only by selecting an alternative config (to config.plist) as from above, the DisabledAML was ignored. In my case it kept loading <string>SSDT-UIAC.aml</string> and I could see it on IORegistryExplorer (as I did not want this to happen so I can use Hackintool, for example).

The method used was:
- Boot to Clover screen
- Select options
- Select an alternative config file
- Go back to desired drive (i.e. partition) to boot
- Press "Enter"

No further actions were done after selecting an alternative config that may have resulted to my findings... Thanks again

asava, 2020-05-01:
Hello,
I hope I addressed it with last commit. Please test.

asava, 2020-05-05:
As no further comments received, I assume fixed.

closed 2020-05-05 by asava

mackonsti, 2020-05-05:
Hi @asava I am sorry, due to the current situation I was not able to test at all and not sure if I would risk updating Clover with all the previous problems reported since the stable (for me) r5107. Thank you for your work and for reviewing the code; allow me to come back after I update Clover again. Many thanks.