CloverHackyColor/CloverBootloader #108: clover v5.0 r5108 failed installation

MatsOdB, 2020-03-30:
I have tried to install clover a couple times now but I always get the same error message.
I have seen the closed issue white the same problem yet I did not find any way to solve this problem.

**### Error code:**
Checking files...
Installation begin..

0:143: execution error: Clover Installer r5108 (v1.1.2), 2020-03-30 12:53:56
macOS Version 10.15.2 (Build 19C57)
SELF = /Applications/Clover.app/Contents/MacOS/CloverDaemonNew
bootSectorsInstallSrc = /Applications/Clover.app/Contents/SharedSupport/bootsectors-install
Target volume: /Volumes/EFI 1
BSD Name:      disk2s1
Filesystem:    FAT32
Is internal:   true
Is writable:   true
o_Ops: '/Volumes/EFI 1' appear to be not writable, remounting read/write.
sleeping 6 seconds..
Error: target volume "/Volumes/EFI 1" is not writable. (1)

Installation failed.

vectorsigma72, 2020-03-30:
Hi, can you by chance see if on another ESP, the problem is the same?

SergeySlice, 2020-04-06:
Wrong configuration. It looks like double EFI volumes and remaining link to absent volume.

closed 2020-04-06 by SergeySlice