CloverHackyColor/CloverBootloader #109: Time out not working

lucosi, 2020-03-30:
After the update from Clover v5.0 r5004 to Clover v5.0 r5107(Mouse and time out not working) and r5108(Time out still not working), the time out stops working.
Same computer and "config.plist" working before. Have you guys any suggestions for me to try? Thank you for the amazing work you guys are doing.

mackonsti, 2020-03-31:
Hi @lucosi not sure how you had time-out working before or what your working `config.plist` has as parameters, but mine stopped working long time ago. After searching for a solution, the only way to have it working on my haxie is by setting the **exact name** of the boot device in DefaultVolume:

```
	<key>Boot</key>
	<dict>
		<key>Arguments</key>
		<string>-igfxnohdmi</string>
		<key>Debug</key>
		<false/>
		<key>DefaultVolume</key>
		<string>Samsung NVMe</string>
		<key>HibernationFixup</key>
		<false/>
		<key>NeverHibernate</key>
		<true/>
		<key>NoEarlyProgress</key>
		<true/>
		<key>Secure</key>
		<false/>
		<key>Timeout</key>
		<integer>10</integer>
		<key>XMPDetection</key>
		<string>Yes</string>
	</dict>
```
At the same time, my GUI has these Scan entries:

```
	<key>GUI</key>
	<dict>
		<key>Hide</key>
		<array>
			<string>Preboot</string>
			<string>Recovery</string>
		</array>
		<key>Mouse</key>
		<dict>
			<key>Enabled</key>
			<false/>
		</dict>
		<key>Scan</key>
		<dict>
			<key>Entries</key>
			<true/>
			<key>Legacy</key>
			<false/>
			<key>Tool</key>
			<true/>
		</dict>
		<key>Theme</key>
		<string>DarkBoot</string>
	</dict>
```

Perhaps you can try DefaultVolume and Scan parameters as above, with the hope it works for you. Ciao, rimanga sicuro

lucosi, 2020-03-31:
Thank you @mackonsti you are the best!
 1 week looking for the solution and now the times out are working.

mackonsti, 2020-03-31:
You are lucky, as in October I searched forums for many days :D I am happy it worked for you!

vectorsigma72, 2020-04-03:
> You are lucky, as in October I searched forums for many days :D I am happy it worked for you!

Boot->DefaultVolume must be a valid  entry. Maybe
`Boot->DefaultVolume = LastBootedVolume`
solve the issue?

mackonsti, 2020-04-04:
Hi @vectorsigma72 setting the value `Boot->DefaultVolume = LastBootedVolume` stopped having time-out working for me since a few months, I think it stopped since the last r4xxx releases. I read in a forum where multiple people mentioned that the time-out stopped working, that if you define now the bootable partition, then the timeout re-works, and so it did in my case.

SergeySlice, 2020-04-06:
Same as issue #111.

SergeySlice, 2020-04-19:
It works.

closed 2020-04-19 by SergeySlice