CloverHackyColor/CloverBootloader #110: r5108: Blackscreen when booting Catalina 10.15.3 with NVidia Graphics

TonyC5, 2020-04-04:
I have Catalina 10.15.3 running without issues on a Dell Latitude E6410 after installing Catalina with DosDude Catalina patcher.  The Catalina installation and subsequent operation of Catalina are without issues when using CLOVER bootloader R5103.

After upgrading to CLOVER r5108, booting Catalina results in blackscreen.  Reverting to R5103 restores normal graphics operation.  My configuration uses CLOVER's "Inject NVidia" graphics option.

TonyC5, 2020-04-05:
Not certain, but I suspect that [this commit ](https://github.com/CloverHackyColor/CloverBootloader/commit/e149dafe4d198ca557b85e95e08be21b844dd0e3) since r5108 is intended to address this problem.

SergeySlice, 2020-04-06:
This is the commit that fix the issue.

closed 2020-04-06 by SergeySlice

TonyC5, 2020-04-06:
@SergeySlice - Thank you!