CloverHackyColor/CloverBootloader #111: Default boot volume is not working

vothaianh, 2020-04-05:
Hi guys, after upgrade to Clover 5107 and also tried 5108, my Clover boot screen stay at volume choose even I set timeout and default volume is "macOS" (it's my mac volume name)

Here my boot log:

6:913  0:003   Found Clover
6:917  0:003  EmuVariable UninstallEmulation: CloseEvent = Success, original var services restored
6:928  0:011  EmuVariable InstallEmulation: orig vars copied, emu.var.services installed, CreateEvent VirtualAddressChange = Success, CreateEvent ExitBootServices = Success, set Status=Success
6:970  0:042  === [ GetEfiBootDeviceFromNvram ] ===============
6:973  0:002  Default boot entry not found
6:976  0:003  EmuVariable UninstallEmulation: CloseEvent = Success, original var services restored
6:986  0:009  DefaultIndex=-1 and MainMenu.Entries.size()=12
7:559  0:572  GUI ready

SergeySlice, 2020-04-06:
Boot to macOS, open System Preferences->Startup Disk, chose your OS to boot and click Restart.
After successful restart you will have default boot by timeout.
The issue is because NVRAM variable is in priority on config.plist setting.

SergeySlice, 2020-04-06:
Same as issue 109

closed 2020-04-06 by SergeySlice