CloverHackyColor/CloverBootloader #112: r5109 verbose boot not working

MysticalOS, 2020-04-06:
with r5103 and lower, verbose boot was fine with -v but now after installing 5109 (skipped 5104-5108 because I wanted to avoid beta testing the C++ rewrite code til it was more stable)

Now, regardless of verbose boot arg, it always boots in apple logo only.

BlvckBytes, 2020-04-06:
Same for me, mine actually gets stuck, LOL.
First of all, my theme is completely corrupted (forgot to make a photo before reverting, sry...), and after selecting OSX to boot, the row of plus symbols appears top left and the apple logo shows, but the verbose output just does not make it to my screen.

Looks like the frame memory from the plus row just gets the apple logo added, and now I ended up with a merge. Going to try to re-install clover, but I think this wont change much, this just does not seem to work with my system.

BlvckBytes, 2020-04-06:
Just tried to reinstall clover, same results (or even worse): Now the boot doesn't work at all. Photos included. When reverting to r5108 everything works fine, tho.

[image]

[image]

mackonsti, 2020-04-07:
I would like to also confirm that, just updated from r5107 to r5109 and I cannot boot in Single mode, the Verbose mode shows the Clover output over the Apple boot screen and Themes are not taken into consideration from config.plist

LeeBinder, 2020-04-07:
ditto

makito89, 2020-04-09:
I also have the same issue.
I've updated from 5108 to 5109 and the only difference between without and with verbose is that with verbose active I see only the text output from loading kernel and after that it switches to graphical mode and I see the Apple logo with progress bar but the text from kernel load still present on screen until framebuffer gets initialized.

makito89, 2020-04-11:
I've tested with the build d74b431 that I found on a post on InsanelyMac "Clover General discussion" thread and the issue with verbose is gone.
https://www.insanelymac.com/forum/applications/core/interface/file/attachment.php?id=346456

corint1, 2020-04-11:
> I've tested with the build [d74b431](https://github.com/CloverHackyColor/CloverBootloader/commit/d74b431b6fb7729a8a484211ef79582d10730f17) that I found on a post on InsanelyMac "Clover General [...]

issue is here ... no change anything

makito89, 2020-04-11:
just updated to r5110 and I can confirm that the issue is still present
[image]

SergeySlice, 2020-04-11:
all questions about 5109 closed

closed 2020-04-11 by SergeySlice