CloverHackyColor/CloverBootloader #113: Minimalism theme broken (r5109)

pit512, 2020-04-06:
New Clover GUI issue with 5109...
This theme is supposed to be white on black.

[image]

CMMChris, 2020-04-07:
Catalina Theme that I use looks screwed up as well.

mifjpn, 2020-04-07:
Hello pit512
me too.
Brue-backscreen is on the screen at minimalinsm & BGM256.

vectorsigma72, 2020-04-07:
minimalism blue, green or orange?

mifjpn, 2020-04-07:
A little while ago, I launched a thread on this situation in a Japanese forum.
I was surprised at the user's response in the forum.
I was told that Minimalism2 that I made had no problems.
I tried it and it was fine.
If you like, please try it.
I will link again.
[minimalism2.zip](http://mifmif.mydns.jp/pcpc/public/minimalism2.zip)

pit512, 2020-04-07:
> minimalism blue, green or orange?

Black

aptonline, 2020-04-07:
Can also confirm 'embedded' theme the same blue background as opposed to black.

pit512, 2020-04-08:
Look! Many themes are broken, even the vector ones...

Christmas theme
[image]

Clovy theme (vector)
[image]

Embedded theme
[image]

Newyear theme
[image]

BrushXue, 2020-04-09:
Same problem here.

SergeySlice, 2020-04-11:
all questions about 5109 closed

closed 2020-04-11 by SergeySlice