CloverHackyColor/CloverBootloader #114: Booting broken from 5108 to 5109

gxsolace, 2020-04-07:
Updated to 5109 and get garbled graphics when booting.
Running Catalina 10.15.4 with Radeon Vega 64.
Was booting just fine on 5108.

gxsolace, 2020-04-07:
Reverting back to 5108, everything boots normally.

florindogarini, 2020-04-07:
Hi, same problem with Radenon VII, EVGA X299 DARK and Catalina 10.15.4.

vectorsigma72, 2020-04-07:
Guys, no one of you can produce a preboot.log?

florindogarini, 2020-04-07:
Hi, I tried with Clovy and it failed anyway.
If it can be useful, I also have a RTX 2080ti disabled via SSDT.
How do I make the preboot.log?

mruepp, 2020-04-08:
Default old Bootcamp theme on PB287Q Monitor gets a blue stripe in the lower 1/3 of the screen. 
I set UIScale to 2 and EFILoginHiDPI to 1
In the Bootlog:
2:788  0:000  SetScreenResolution: 2560x1600 - not found!
2:788  0:000  SetMaxResolution: found best mode 0: 3840x2160
2:838  0:049  Video mode change to mode #0: Success
2:838  0:000   - set

malhal, 2020-04-08:
Same here, r5109 blue clover screen then no mouse/keyboard its like login screen is frozen and OS unusable. Lucky had USB backup of r5108 on hand.

Lucke77, 2020-04-09:
Same here..., 5108 icons on catalina4k are ok, on 5109 are set wrong... reverting back to r5108.

svandive, 2020-04-09:
Booting in to 5109 results in a Not symbol, reverting back to 5107 and booting works again.  What logs can I grab to provide assistance?

gxsolace, 2020-04-09:
I notice that USB doesn't work with 5109 as well. No keyboard/mouse.  Attaching directory list of files used.

[image]

malhal, 2020-04-09:
> I notice that USB doesn't work with 5109 as well. No keyboard/mouse. Attaching directory list of files used.
> 
> [image]

Same here but I thought the screen was frozen, can you remote in?
Shouldn't 5109 be pulled?

gxsolace, 2020-04-09:
> Same here but I thought the screen was frozen, can you remote in?
> Shouldn't 5109 be pulled?

Screen frozen with garbled graphics (UI doesn't load) + No USB. Yeah I can remote in using another Mac. Did that and reinstalled 5108 and everything worked fine again.

SergeySlice, 2020-04-11:
all questions about 5109 closed

closed 2020-04-11 by SergeySlice

malhal, 2020-04-11:
I believe this issue still exists, people have just posted the same boot problem with 5110