CloverHackyColor/CloverBootloader #115: New theme engine reports wrong disk logos for windows 10 and ubuntu partitions

jtash, 2020-04-07:
On r5108 and earlier windows and ubuntu partitions gave the correct logos in the clover boot screen. In r5109 they now both show as portable drive logos

Flowwwww, 2020-04-07:
Same problem here. Windows entry show as a mac logo.
Even specifying custom logo path in GUI > Custom Entries doesn't work.

mackonsti, 2020-04-07:
I would like to confirm that too, updated from r5107 to r5109 and the theme also seems totally not taken into consideration.

LeeBinder, 2020-04-07:
me.too

najeebanwer, 2020-04-08:
Yes, here is a selection of what the themes look like with r5109. They all looked a lot better with r5108.
Catalina:
[image]
Mojave:
[image]
Retina:
[image]

SergeySlice, 2020-04-13:
Test 5111

closed 2020-04-13 by SergeySlice