CloverHackyColor/CloverBootloader #117: r5108 and r5109 pkg installer fails - Directory not empty

itsjimbo, 2020-04-08:
Currently using clover r5104
```
tail -1000 /var/log/install.log
```

```
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Executing script "./preinstall" in /private/tmp/PKInstallSandbox.2s9jSu/Scripts/org.clover.drivers64uefi.apfsdriverloader.uefi.TTQX6P
2020-04-08 13:33:01-07 kps-iMac installd[458]: ./preinstall: preinstall: Path to installer....... /Users/kp/Desktop/Clover_r5109.pkg
2020-04-08 13:33:01-07 kps-iMac installd[458]: ./preinstall: preinstall: Path to destination..... /EFIROOTDIR/EFI/CLOVER/drivers/UEFI
2020-04-08 13:33:01-07 kps-iMac installd[458]: ./preinstall: preinstall: Path to dest volume..... /
2020-04-08 13:33:01-07 kps-iMac installd[458]: ./preinstall: preinstall: Root of system folder... /
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Using trashcan path /var/folders/zz/zyxvpxvq6csfxvn_n0000000000000/T/PKInstallSandboxTrash/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.sandboxTrash for sandbox /Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Shoving /Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root (2 items) to /
2020-04-08 13:33:01-07 kps-iMac shove[2688]: Cross device symlink found in shove path, including /usr/standalone/i386/boot1h2
2020-04-08 13:33:01-07 kps-iMac shove[2688]: [source=file] failed _RelinkFile(/Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root/EFIROOTDIR/EFI/BOOT/BOOTX64.efi, /EFI/BOOT/BOOTX64.efi): Directory not empty
2020-04-08 13:33:01-07 kps-iMac shove[2688]: srcPath = /Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root/EFIROOTDIR/EFI/BOOT/BOOTX64.efi NSFileOwnerAccountID=0 NSFileHFSTypeCode=0 NSFileSystemFileNumber=4091488 NSFileExtensionHidden=0 NSFileSystemNumber=16777223 NSFileSize=922944 NSFileGroupOwnerAccountID=0 NSFileHFSCreatorCode=0 NSFileCreationDate=2020-04-06 17:27:46 +0000 NSFileOwnerAccountName=root NSFilePosixPermissions=420 NSFileType=NSFileTypeRegular NSFileGroupOwnerAccountName=wheel NSFileReferenceCount=1 NSFileModificationDate=2020-04-06 17:27:46 +0000
2020-04-08 13:33:01-07 kps-iMac shove[2688]: dstPath = /EFI/BOOT/BOOTX64.efi NSFileOwnerAccountID=0 NSFileHFSTypeCode=0 NSFileSystemFileNumber=1152921500312398766 NSFileExtensionHidden=0 NSFileSystemNumber=16777223 NSFileSize=922944 NSFileGroupOwnerAccountID=0 NSFileHFSCreatorCode=0 NSFileCreationDate=2020-04-06 17:27:46 +0000 NSFileOwnerAccountName=root NSFilePosixPermissions=420 NSFileType=NSFileTypeRegular NSFileGroupOwnerAccountName=wheel NSFileReferenceCount=1 NSFileModificationDate=2020-04-06 17:27:46 +0000
2020-04-08 13:33:01-07 kps-iMac shove[2688]: dstParentPath = /EFI/BOOT NSFileOwnerAccountID=0 NSFileSystemFileNumber=1152921500312397585 NSFileExtensionHidden=0 NSFileSystemNumber=16777223 NSFileSize=96 NSFileGroupOwnerAccountID=0 NSFileOwnerAccountName=root NSFileCreationDate=2020-03-30 09:22:36 +0000 NSFilePosixPermissions=493 NSFileType=NSFileTypeDirectory NSFileGroupOwnerAccountName=wheel NSFileReferenceCount=3 NSFileModificationDate=2020-04-08 20:32:21 +0000
2020-04-08 13:33:01-07 kps-iMac install_monitor[2600]: Re-included: /Applications, /Library, /System, /bin, /private, /sbin, /usr
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: releasing backupd
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: allow user idle system sleep
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Cleared responsibility for install from 2582.
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Cleared permissions on Installer.app
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=120 "An unexpected error occurred while moving files to the final destination." UserInfo={NSLocalizedDescription=An unexpected error occurred while moving files to the final destination., arguments=(
	    "-f",
	    "-s",
	    "/Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root",
	    "/"
	), NSUnderlyingError=0x7f8b2f5ce230 {Error Domain=NSPOSIXErrorDomain Code=66 "Directory not empty"}} {
	    NSLocalizedDescription = "An unexpected error occurred while moving files to the final destination.";
	    NSUnderlyingError = "Error Domain=NSPOSIXErrorDomain Code=66 \"Directory not empty\"";
	    arguments =     (
	        "-f",
	        "-s",
	        "/Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root",
	        "/"
	    );
	}
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Running idle tasks
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Removing client PKInstallDaemonClient pid=2582, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=120 "An unexpected error occurred while moving files to the final destination." UserInfo={NSUnderlyingError=0x600000962d90 {Error Domain=NSPOSIXErrorDomain Code=66 "Directory not empty"}, NSLocalizedDescription=An unexpected error occurred while moving files to the final destination., arguments=(
	    "-f",
	    "-s",
	    "/Library/InstallerSandboxes/.PKInstallSandboxManager/4C153255-4D9F-4856-84F8-0CB16E5DC7D6.activeSandbox/Root",
	    "/"
	)}
2020-04-08 13:33:01-07 kps-iMac installd[458]: PackageKit: Done with sandbox removals
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: PackageKit: Bypassing the enforcement checker since the destination is read/write
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: Install failed: The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: IFDInstallController 7C06C0 state = 8
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: Displaying 'Install Failed' UI.
2020-04-08 13:33:01-07 kps-iMac Installer[2582]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.

```

SergeySlice, 2020-04-13:
Test recent Clover

itsjimbo, 2020-04-14:
r5111 (c572e50ec) also failed - same message

do you know what is the directory that its complaining about not being empty?
Is it  /EFI/BOOT/ or some other install dir?  
```
 /EFI/BOOT/BOOTX64.efi): Directory not empty
```
This setup is pretty basic with on backup and network volumes

```
/dev/disk1s5 on / (apfs, local, journaled)
/dev/disk1s1 on /System/Volumes/Data (apfs, local, journaled, nobrowse)
/dev/disk2s2 on /Volumes/MacintoshBackup (hfs, local, journaled)
```
Never seen this before, perhaps apfs or permissions.. any ideas?

SergeySlice, 2020-04-14:
Test Clover.app in release 5112.

itsjimbo, 2020-04-14:
Success!! 🎉

closed 2020-04-14 by itsjimbo