CloverHackyColor/CloverBootloader #120: 5109 does not boot - black screen first, boot.efi error then.

BbIKTOP, 2020-04-10:
Updated 5108 to 5109 by copying bootx64.efi and cloverx84.efi. 
Now system does not show log at boot with the "-v" boot argument, and when it comes to switch to the gui, I observe black screen. I'm able to login via "screen sharing" then, and observe really strange display mode, like 640x480 or something alike. But display remains black. When I reboot via screen sharing session, system does not boot anymore with boot.efi error 

    [EB]`P:DCCPR] Err(0x0E) @ GV AAPL,CoProcessorReboot
    [EB]`P:DCMOSP] Err(0x0E) @ GV AAPL,PanicInfo0000 (0)

After rollback to the 5108 by replacing both clover's efi files back, system does not boot wit Err(0x0E) opening prelinked kernel.
After rollback, booting to recovery and rebuilding prelinked kernel with kextcache -i /Volumes/Macintosh\HD (in my case) everything works fine.

Also, no themes are working with this release.

Heimatplan, 2020-04-10:
Same error, rolled back to 5103. I got the RX5700XT as well.
Booting without agdpmod=pikera leads to black screen as well with this GPU.

anhvu-sg, 2020-04-11:
Same error (Z390I wifi pro, i9900k, RX590) with r510, but I tried r5110 -> work well

anhvu-sg, 2020-04-11:
Seems the release of r5110 is not stable yet, I shut down my pc and back again, it does not work :( but when I restart PC everything is fine. I decide to come back r5108

BbIKTOP, 2020-04-11:
5108 is stable, at least for the time I run it (from the day it has been released). Thank you, will stay with it till Sergey will fix it.

SergeySlice, 2020-04-11:
all questions about 5109 closed

closed 2020-04-11 by SergeySlice