CloverHackyColor/CloverBootloader #121: 5110 Theme rendering errors Minimalism theme

shrubberies, 2020-04-11:
5110 shows different render errors than 5109 with Mininmalism (black) theme. Blue bar persists at bottom of screen. Background image is not using bg from theme but a larger background image with a clover on it.

shrubberies, 2020-04-11:
looks like its using the bg from the embedded theme but tiled and not scaled,

GCPDetective, 2020-04-11:
I’m experiencing the same issue except it’s with the OSS theme. I’ve tried a few other themes and still get the same thing.

juliantoledo, 2020-04-11:
I think images are not scaling well, I have the same issue in 5109. I had to copy a different background image from another theme.

mifjpn, 2020-04-11:
Me too.
The background doesn't seem to scale well.
I use BGM256 theme. I matched the background PNG file to the display resolution.
Then, it goes well.

theliamllyd, 2020-04-11:
Having issues here with 5109 / 5110. Just displays as a blue background.

[image]

BrushXue, 2020-04-11:
Still same issue with r5110. Resolution 3840*2160
[image]

sryze, 2020-04-12:
I have a similar problem with the BGM theme after updating to 5110:

[image]

SergeySlice, 2020-04-13:
Resolved

closed 2020-04-13 by SergeySlice