CloverHackyColor/CloverBootloader #122: consecutive 5109.  5110 boot USB freeze and more errors on High Sierra

vacarogit, 2020-04-11:
boot Intel USB freeze, thus KB, mouse not accessible (have to hook mouse to PXSX), plz don't release before validation.

danieldabate, 2020-04-11:
I'm having the same issue, rollbacking to 5108 for now.

vacarogit, 2020-04-11:
> I'm having the same issue, rollbacking to 5108 for now.

we need qty control

SergeySlice, 2020-04-13:
The issue about USB or about validation mechanism?

danieldabate, 2020-04-13:
Originally it was about the USB freeze, but @vacarogit changed the title. I don't think the new title helps to understand the real problem.

vacarogit, 2020-04-16:
> The issue about USB or about validation mechanism?

it's about validation in the root, because of that since 5008, multiple version, 5009 n 5011 both have boot issue, cause inconvenience.

SergeySlice, 2020-04-19:
If someone will develop new project somehow validation of you USB sticks then he will open new github project.
I see no what I shall do with Clover.

closed 2020-04-19 by SergeySlice