CloverHackyColor/CloverBootloader #125: Clover builds past 5099 break booting. Catalina 10.15.3

acdop100, 2020-04-11:
Hey there,

I have repeatedly tried to update my clover install over the past few months with no success. Every time I try and (verbose) boot into my drive after an update, it shows the apple logo with a line of plus's (++++++...) at the top and then nothing happens. Same thing in safe mode. Does anyone know what could be causing this? All of my kexts are up to date and I updated my extra files/drivers from within the clover installer too. 

I can provide debug logs or more if you need them. Let me know. Thanks!

vacarogit, 2020-04-12:
You don’t “read” previous comments don’t you?

acdop100, 2020-04-12:
> You don’t “read” previous comments don’t you?

What do you mean? It doesn't look like there are any other issues that are directly related to this from the past 4 months or so.

SergeySlice, 2020-04-12:
You should provede a little more information if this is a Clover issue and not user.

acdop100, 2020-04-12:
> You should provide a little more information if this is a Clover issue and not user mistake.

I understand. What would be most helpful for you to help me? I will attach my config.plist to the top.

vectorsigma72, 2020-04-12:
> > You should provide a little more information if this is a Clover issue and not user mistake.
> 
> I understand. What would be most helpful for you to help me? I will attach my config.plist to the top.

a preboot.log will be also nice

SergeySlice, 2020-04-19:
The issue tracker is not for helping new users.
Opening an issue you have to provide a proof what you say.

closed 2020-04-19 by SergeySlice