CloverHackyColor/CloverBootloader #129: New theme engine issue (after rev. 5109)

forumi0721, 2020-04-12:
Hi,
After 5109, Clover cannot boot.
Please check blow message.

9:633 0:61 XArray::ElementAt(xsize) -> operator [] : index > m_lenA fatal error happened. System halted.

My Spec
Clover Version : 5108 (Latest working version)
macOS Version : 10.15.4
Clover Theme : mimi black (https://sourceforge.net/p/cloverefiboot/themes/ci/master/tree/themes/mimi%20black/)
ETC : I use OEM path for specific board (OEM/Z77 Extreme4/config.plist)

richtw370, 2020-04-12:
behavior seems to vary based on theme selected.  I have received that error in 5019, 5110 and 5111 using BGM theme.  for other themes I have seen the background panelled rather than spread across screen.  I found that the Carbon theme does work without issue and for now am using that

pagimamaun, 2020-04-13:
@richtw370 i tried different background carbon included and background stay always until resolution change. for exemple when selecting linux grub did not appear.

malhal, 2020-04-13:
> behavior seems to vary based on theme selected. I have received that error in 5109, 5110 and 5111 using BGM theme. for other themes I have seen the background panelled rather than spread across screen. I found that the Carbon theme does work without issue and for now am using that

So the Carbon theme worked but could you actually login to macOS or does it still freeze at login with USB broken?

richtw370, 2020-04-13:
for me, using 5111, on my nuc8, clover boots up and macOS starts fine...no issue

SergeySlice, 2020-04-13:
Test 5111

closed 2020-04-13 by SergeySlice

forumi0721, 2020-04-14:
I fix theme, bg.png is problem.
I copy other transparent png image to background, it's OK. but other issue remains.

asava, 2020-05-01:
Should be fully fixed in 5115. If anything persists, please write.

malhal, 2020-05-11:
Experiencing 1 core 100% kernel_task on 5116 still testing though.