CloverHackyColor/CloverBootloader #132: Clover r5111 still has problems with some themes

najeebanwer, 2020-04-13:
I am still having an issue with the Catalina theme after installing r5111 (see the blue region at the bottom of the picture below). Unless this is a feature and not a bug? My screen resolution is 1920x1200. This used to work correctly in r5108.
[image]

pit512, 2020-04-13:
I managed to completely fill the Minimalism theme background by changing the type from _Scale_ to _Tile_...

`		<key>Background</key>`
`		<dict>`
`			<key>Type</key>`
`			<string>Tile</string>`
`		</dict>`

It seems newest Clover can only scale the background horizontally!
Hope it helps...

najeebanwer, 2020-04-17:
This appears to be fixed in Clover r5112. See screenshots below.
Catalina theme:
[image]
Retina theme:
[image]

closed 2020-04-17 by najeebanwer