CloverHackyColor/CloverBootloader #136: Theme issue with BootCampStyle and Indicator

forumi0721, 2020-04-14:
Hi,
After 5109(New theme engine), BootCampStyle indicator does not works correctly.

When i have 2 (or more) boot entry, indicator works but not clear when move cursor.
Always remains indicator.

I tested some theme (mac_s, mimi black, etc..), always remain indicator likes multiple selection.

Please check and fix this issue.

Thanks.

allegrox, 2020-04-16:
> Hi,
> After 5109(New theme engine), BootCampStyle indicator does not works correctly.
>
> When i have 2 (or more) boot entry, indicator works but not clear when move cursor.
> Always remains indicator.
>
> I [...]

I had the same problem

asava, 2020-04-18:
Please try with current changes (from source), see if issue is resolved.

closed 2020-04-22 by asava