CloverHackyColor/CloverBootloader #137: Bluetooth problem

OvidiuNetcu, 2020-04-15:
From version Release v5.0 r5107, all version have bluetooth problem. Is lost conection and is not work to pair mous and keyboards over buetooth.
I use a airport from a Imac 2014. Is work very well with version r5107.

OvidiuNetcu, 2020-04-15:
0:100  0:100  MemLog inited, TSC freq: 3600484080
0:100  0:000  CPU was calibrated with RTC
0:100  0:000  
0:100  0:000  Now is 14.02.2020,  19:10:02 (GMT)
0:100  0:000  Starting Clover revision: 5103 (master, commit 3f3b85da0) on American Megatrends EFI
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED --conf=/Users/sergey/src/CloverHackyColor/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SelfDevicePath=PciRoot(0x0)\Pci(0x1D,0x0)\Pci(0x0,0x0)\NVMe(0x1,31-31-20-31-29-A7-79-64)\HD(1,GPT,10CD5D8B-83A4-4DE2-ADBA-98A16D93EBE4,0x28,0x64000) @358A2C18
0:100  0:000  SelfDirPath = \EFI\BOOT
0:100  0:000  SimpleTextEx Status=Success
0:100  0:000  === [ Get Smbios ] ========================================
0:100  0:000  Type 16 Index = 0
0:100  0:000  Total Memory Slots Count = 4
0:100  0:000  Type 17 Index = 0
0:100  0:000  SmbiosTable.Type17->Speed = 3000MHz
0:100  0:000  SmbiosTable.Type17->Size = 8192MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 ChannelA-DIMM0
0:100  0:000  SmbiosTable.Type17->Vendor = 04CB
0:100  0:000  SmbiosTable.Type17->SerialNumber = 564D0000
0:100  0:000  SmbiosTable.Type17->PartNumber = DDR4 3000 2OZ       
0:100  0:000  Type 17 Index = 1
0:100  0:000  SmbiosTable.Type17->Speed = 3000MHz
0:100  0:000  SmbiosTable.Type17->Size = 8192MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 1 ChannelA-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = 04CB
0:100  0:000  SmbiosTable.Type17->SerialNumber = 713A0400
0:100  0:000  SmbiosTable.Type17->PartNumber = DDR4 3000 2OZ       
0:100  0:000  Type 17 Index = 2
0:100  0:000  SmbiosTable.Type17->Speed = 3000MHz
0:100  0:000  SmbiosTable.Type17->Size = 8192MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 2 ChannelB-DIMM0
0:100  0:000  SmbiosTable.Type17->Vendor = 04CB
0:100  0:000  SmbiosTable.Type17->SerialNumber = 90320400
0:100  0:000  SmbiosTable.Type17->PartNumber = DDR4 3000 2OZ       
0:100  0:000  Type 17 Index = 3
0:100  0:000  SmbiosTable.Type17->Speed = 3000MHz
0:100  0:000  SmbiosTable.Type17->Size = 8192MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 3 ChannelB-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = 04CB
0:100  0:000  SmbiosTable.Type17->SerialNumber = 5A4D0000
0:100  0:000  SmbiosTable.Type17->PartNumber = DDR4 3000 2OZ       
0:100  0:000  Boot status=0
0:100  0:000  Running on: 'Z390 AORUS PRO' with board 'Z390 AORUS PRO-CF'
0:100  0:000  === [ GetCPUProperties ] ==================================
0:100  0:000  CPU Vendor = 756E6547 Model=906ED
0:100  0:000   The CPU supported SSE4.1
0:100  0:000  BrandString = Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz
0:100  0:000   The CPU supported turbo
0:100  0:000  MSR 0x35               80008
0:100  0:000   TSC/CCC Information Leaf:
0:100  0:000    numerator     : 300
0:100  0:000    denominator   : 2
0:100  0:000   Calibrated ARTFrequency: 24003227
0:100  0:000   Rounded ARTFrequency: 24000000
0:100  0:000  MSR 0xE2 before patch 1E008000
0:100  0:000  MSR 0xE2 is locked, PM patches will be turned on
0:100  0:000  MSR 0xCE              08080838_F1012400
0:100  0:000  corrected FLEX_RATIO = E0000
0:100  0:000  FSBFrequency = 127 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:100  0:000  MaxDiv/MinDiv: 36.0/8
0:100  0:000  Turbo: 47/48/48/49
0:100  0:000  Features: 0xBFEBFBFF
0:100  0:000  Threads: 8
0:100  0:000  Cores: 8
0:100  0:000  FSB: 100 MHz
0:100  0:000  CPU: 4554 MHz
0:100  0:000  TSC: 4554 MHz
0:100  0:000  PIS: 100 MHz
0:100  0:000  ExternalClock: 25 MHz
0:100  0:000  === [ GetDevices ] ========================================
0:100  0:000  PCI (00|00:00.00) : 8086 3E30 class=060000
[... 210 lines cut ...]
6:642  0:005          AddLoaderEntry for Volume Name=EFI
6:642  0:000          skipped because entry is hidden
6:642  0:000  - [06]: 'Preboot', hidden
6:642  0:000  - [07]: 'Machintosh - Data'
6:643  0:000  - [08]: 'Recovery', hidden
6:643  0:000  - [09]: 'VM'
6:643  0:000  - [10]: 'Machintosh'
6:648  0:004          AddLoaderEntry for Volume Name=Machintosh
6:650  0:001      Check if volume Is Hibernated:
6:650  0:000      Check sleep image 'by signature':
6:650  0:000      using default sleep image name = \private\var\vm\sleepimage
6:650  0:000      sleepimage not found -> Not Found
6:650  0:000      hibernated: no - sign
6:651  0:001  === [ AddCustomTool ] =====================================
6:652  0:000  found tool \EFI\CLOVER\tools\Shell64U.efi
6:652  0:000  Checking EFI partition Volume 1 for Clover
6:652  0:000  Checking EFI partition Volume 4 for Clover
6:652  0:000   Found Clover
6:652  0:000  EmuVariable UninstallEmulation: CloseEvent = Success, original var services restored
6:653  0:000  EmuVariable InstallEmulation: orig vars copied, emu.var.services installed, CreateEvent VirtualAddressChange = Success, CreateEvent ExitBootServices = Success, set Status=Success
6:673  0:020  === [ GetEfiBootDeviceFromNvram ] =========================
6:673  0:000   - found entry 0. 'Boot macOS from Machintosh', Volume 'Machintosh', DevicePath 'PciRoot(0x0)\Pci(0x1D,0x0)\Pci(0x0,0x0)\NVMe(0x1,31-31-20-31-29-A7-79-64)\HD(2,GPT,7AF64392-48A7-4385-8FBE-86692E114EFB,0x64028,0x746A2D60)\VenMedia(BE74FCF7-0B7C-49F3-9147-01F4042E6842,701184A8B638F4498C781DF7EF55993B)\System\Library\CoreServices\boot.efii'
6:673  0:000  EmuVariable UninstallEmulation: CloseEvent = Success, original var services restored
6:673  0:000  DefaultIndex=0 and MainMenu.EntryCount=7
7:173  0:500  GUI ready

SergeySlice, 2020-04-15:
All is good here.

asava, 2020-04-17:
There was an error that could have affected it. Please try again with latest sources.

OvidiuNetcu, 2020-04-18:
I try and ai notice

OvidiuNetcu, 2020-04-18:
Install last version, test on start up and sleep. Its ok now. Thx

closed 2020-04-18 by OvidiuNetcu