CloverHackyColor/CloverBootloader #138: Clover ignores CustomEDID injecting

hectorox, 2020-04-15:
Hi,

I am so frustrated here, tried almost everything for two days, but clover (r5111) is not injecting my customEDID values (clover seems to ingore it). In several forums I heard this was a bug in the clover bootloader itself. Can this be fixed?

SergeySlice, 2020-04-15:
How do you know if clover does not injected EDID? Log? System dumps?

hectorox, 2020-04-15:
because I do not get any results, when I run in terminal:
ioreg -l | grep "I0DisplayEDID"
[hectorox’s MacBook Pro.ioreg.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4483560/hectorox.s.MacBook.Pro.ioreg.zip)
you can take a look into my ioreg report, let me know, if you need anything else

vectorsigma72, 2020-04-15:
> because I do not get any results, when I run in terminal:
> ioreg -l | grep "I0DisplayEDID"

maybe `ioreg -l | grep IODisplayEDID`  with the O (letter) instead of 0 (zero) could work better?

hectorox, 2020-04-15:
Sorry, that was a typo on my side. I actually used O instead of 0, stil no result (you can the the same in the ioreg

vectorsigma72, 2020-04-16:
> Sorry, that was a typo on my side. I actually used O instead of 0, still no result (you can the the same in the ioreg)

Then give a try to r5112 and post the boot log. May be Slice can found it more usefull

hectorox, 2020-04-16:
I have installed clover r5112 now, but it is still the same :(
I have attached the boot log
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/4485933/preboot.log)

vectorsigma72, 2020-04-16:
> I have installed clover r5112 now, but it is still the same :(
> I have attached the boot log
> [preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/4485933/preboot.log)

looks like a protocol is missing that I think is available in CsmVideoDxe.efi. But, your laptop didn't work better in UEFI mode?

hectorox, 2020-04-16:
I would love to use the UEFI mode, but this notebook does not support UEFI.

vectorsigma72, 2020-04-16:
So the question is... what happen if you set Graphics->LoadVBios = true,  instead of false?

hectorox, 2020-04-16:
nothing changed :(
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/4488064/preboot.log)

SergeySlice, 2020-04-19:
> because I do not get any results, when I run in terminal:
> ioreg -l | grep "I0DisplayEDID"
> and because the DisplayproductID and VendorID is still the same and not the ones from the CustomEDID values I have patched
> [hectorox’s MacBook Pro.ioreg.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4483560/hectorox.s.MacBook.Pro.ioreg.zip)
> you can take a look into my ioreg report, let me know, if you need anything else

Is it the EDID you injected?
~~~
<00 ff ff ff ff ff ff 00 06 10 f2 9c 00 00 00 00 1a 15 01 04 95 1a 0e 78 02 ef 05 97 57 54 92 27 22 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 29 36 80 a0 70 38 1f 40 30 20 25 00 61 c6 10 00 00 19 00 00 00 fc 00 43 6f 6c 6f 72 20 4c 43 44 0a 20 20 20 00 00 00 fe 00 53 41 4d 53 55 4e 47 0a 20 20 20 20 20 00 00 00 fe 00 31 36 30 48 54 30 31 2d 41 30 32 0a 20 00 3d>
~~~

It is the property AAPL00,override-no-connect. It means the EDID will be used if and only if it can't be obtained by I2C/DDC between graphics card and monitor.
Else native EDID will be used so why you see no your patches.

closed 2020-04-19 by SergeySlice

hectorox, 2020-04-19:
Hi Sergej,

Yes, this is the EDID code that I need to inject to make the panel recognisable for OS X. If I understand you correctly, CustomEDID will only get pushed, when the communication between graphic card and monitor is not working correctly? Strange, I am using on a second device in UEFI Mode also customEDID with clover (an old build from Rehabman) and the injection is working fine there and I am 100% sure the communication between graphic card and monitor is working fine. I only changed the EDID so that the monitor only supports one mode displaying and not two modes. So, is there a way to force a customEDID? That would really help me a lot