CloverHackyColor/CloverBootloader #142: Default boot volume not being observed

norton287, 2020-04-20:
Instead of the boot volume in the config.plist it just selects the first volume as its choice to boot.  I have double-checked my plist and tried several variations on different bootable volumes but it always defaults to the first volume in the list on the clover menu.

SergeySlice, 2020-04-20:
It is done by NVRAM.
You have to open System Preferences -> Startup Disk -> chose one and click restsrt.

norton287, 2020-04-20:
That doesn’t work either, it still selects the first disk in the list in the clover menu.

SergeySlice, 2020-04-20:
Type F2, get EFI/CLOVER/misc/preboot.log and show it.
There are information about all your volumes and what clover selected.
I don't understand what you mean. Clover works fine in choosing a volume to boot. It works more then 8 years.

norton287, 2020-04-20:
I can select any volume to boot, it is just not highlighting the default volume to boot according to the config.plist.  it just highlights by default the first volume.

Here is the reboot

norton287, 2020-04-20:
Here is the boot section from my config.plist, Beta should be the default volume selected like it has been for a very long time but instead the volume that appears first in the selection list is selected.

dict>
		<key>Arguments</key>
		<string>dart=0 darkwake=1 -v keepsyms=1 debug=0x100 kext-dev-mode=1</string>
		<key>Debug</key>
		<false/>
		<key>DefaultVolume</key>
		<string>Beta</string>
		<key>Legacy</key>
		<string>PBR</string>
		<key>Secure</key>
		<false/>
		<key>Timeout</key>
		<integer>5</integer>
		<key>XMPDetection</key>
		<integer>1</integer>
	</dict>

norton287, 2020-04-20:
Try this again

`dict>
		<key>Arguments</key>
		<string>dart=0 darkwake=1 -v keepsyms=1 debug=0x100 kext-dev-mode=1</string>
		<key>Debug</key>
		<false/>
		<key>DefaultVolume</key>
		<string>Beta</string>
		<key>Legacy</key>
		<string>PBR</string>
		<key>Secure</key>
		<false/>
		<key>Timeout</key>
		<integer>5</integer>
		<key>XMPDetection</key>
		<integer>1</integer>
	</dict>`

SergeySlice, 2020-04-21:
Setting in NVRAM has a priority on config.plist.

norton287, 2020-04-21:
I tried setting the drive as the boot drive in System Preferences - Startup but I get the error that the NVRAM tool was unable to set a preference when trying that.

norton287, 2020-04-22:
Here's is my nvram.plist, everything appears to be in hex.  Is there a way to change the specialbootdrive entry.  I think that is what is selecting macOS as the default volume.  I need it to be Beta instead.
[nvram.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4517757/nvram.plist.zip)

SergeySlice, 2020-05-06:
Clover works fine in choosing a volume to boot. It works more then 8 years.

closed 2020-05-06 by SergeySlice