CloverHackyColor/CloverBootloader #143: No os_icons match for entries like "UEFI internal"

zty199, 2020-04-21:
Hi,
I just update Clover to 5113. But I find a strange phenomenon.

When I disable all options about os icons in theme.plist, the volume icons show correctly.
However, if I enable "Exchange" option, only entries for OS show their icons correctly, others like "UEFI internal" still show just a small black-and-yellow square.

Is there any icons for this kind of entries? And how should I name them?

Or, when users enable "Exchange" option for os icons, if os_icon which matches the entry exists, the icon would be exhcnaged. Otherwise the origin volume icon should be shown as default?

I don't know how you make this work, and my English is poor. Hope you can understand what I am talking about.

Thank you.

SergeySlice, 2020-04-21:
What is "exchange"?

closed 2020-04-21 by SergeySlice

zty199, 2020-04-22:
Sorry, I mean the option "Swap" in Theme->Badges in theme.plist.

zty199, 2020-04-22:
> What is "exchange"?
Sorry, I mean the option "Swap" in Theme->Badges in theme.plist. When I enable this option, part of entries like "UEFI internal" can't show their icons correctly. I don't know how should I name this kind of icon.