CloverHackyColor/CloverBootloader #145: ACPI/DropTable setting not working (after 5114)

forumi0721, 2020-04-22:
Hi,
After v5114,DropTables setting in config.plist not working.
On GUI ACPI config, not checked, and not working.
But when I check manually, it works.

I check commits, but I found only one commit that relative DropTables is DropForAllOS.
So I set DropForAllOS all tables, but also not working.
d7a38dce3e635d0080e403a54faa0f33254dc0b2

Any other's OK?

Please check
Thanks.

------MY config.plist

		<key>DropTables</key>
		<array>
			<dict>
				<key>DropForAllOS</key>
				<true/>
				<key>Signature</key>
				<string>DMAR</string>
			</dict>
			<dict>
				<key>DropForAllOS</key>
				<true/>
				<key>Signature</key>
				<string>SSDT</string>
				<key>TableId</key>
				<string>CpuPm</string>
			</dict>
			<dict>
				<key>DropForAllOS</key>
				<true/>
				<key>Signature</key>
				<string>SSDT</string>
				<key>TableId</key>
				<string>Cpu0Ist</string>
			</dict>
			<dict>
				<key>DropForAllOS</key>
				<true/>
				<key>Signature</key>
				<string>MCFG</string>
			</dict>
		</array>

asava, 2020-04-22:
DropForAllOS is irrelevant for injecting to MacOS, so you do not need to set it.
Also, the code for that commit is not used at all when booting to OSX.

If you say the you don't have any option tagged in the GUI with the config.plist above, it is likely that have an error in your config.plist, so it cannot be parsed.
If you are sure you have no error in your config.plist, please attach your config.plist and your boot.log to better see what is going on.

asava, 2020-04-22:
Corrected in master. Please test from sources, it should be OK now.

closed 2020-04-24 by asava

forumi0721, 2020-04-26:
Thanks, It works.