CloverHackyColor/CloverBootloader #146: No os_icons match for entries like "UEFI internal"

zty199, 2020-04-23:
I'm sorry for my poor expression in last issue.

I find a strange phenomenon when using Clover 5113.

When I disable all options about os icons in theme.plist, the volume icons show correctly.
However, if I enable "Swap" option in Theme->Badges in theme.plist, only entries for OS show their icons correctly, others like "UEFI internal" still show just a small black-and-yellow square.

Is there any icons for this kind of entries? And how should I name them?

Or, when users enable "Swap" option for os icons, if os_icon which matches the entry exists, the os icon and the vol icon would be swapped. Otherwise the origin volume icon should be shown as default?

I don't know how you make this work, and my English is poor. Hope you can understand what I am talking about.

Thank you.

asava, 2020-04-24:
Confirmed.

asava, 2020-04-25:
Corrected in master. It will be os_unknown for unknown OS and os_clover for EFI partition

closed 2020-04-25 by asava