CloverHackyColor/CloverBootloader #147: r5114 breaks my system

cryptooth, 2020-04-23:
I'm getting "m_lenA fatal error happened. System halted" at boot. And system halted.

malhal, 2020-04-23:
Thanks, still on 5108 waiting for this to be fixed will keep waiting

asava, 2020-04-23:
It's probably already corrected in master. An issue that could cause this was identified and fix. It you can compile from sources, please test.

TonyC5, 2020-04-23:
@asava Can you post a link to the commit that fixes this?  Thank you.

asava, 2020-04-23:
There were two commits, actually: [this](https://github.com/CloverHackyColor/CloverBootloader/commit/f8c1f08c136e0693dc8a1fce544b3419c488af09) and [this](https://github.com/CloverHackyColor/CloverBootloader/commit/978e35845a42a0b6caf6f18eabaf3db3e0482a99)

And if you want a recent binary, you can find it here: [binary](https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2718867)

cryptooth, 2020-04-23:
> It's probably already corrected in master. An issue that could cause this was identified and fix. It you can compile from sources, please test.

Can I update from directly clover menu? I did this operation in the morning. Will it download the corrected version? How can I check build number?

[image]

vacarogit, 2020-04-24:
> Thanks, still in 5108 waiting for this to be fixed will keep waiting

Yes, I confirm that ever Since 5108, I never get any later rev. right without system halt!
What's going wrong, this really needs a Rework based on 5108 back, not newer ones which are already corrupted in codes

asava, 2020-04-24:
> > It's probably already corrected in master. An issue that could cause this was identified and fix. It you can compile from sources, please test.
> 
> Can I update from directly clover menu? I did this operation in the morning. Will it download the corrected version? How can I check build number?
No, that will work only after 5115 is released.

paaboo, 2020-04-26:
> I'm getting "m_lenA fatal error happened. System halted" at boot. And system halted.

Please change other theme in the config.plist file.

asava, 2020-04-29:
Have the issues been resolved with r5115?

cryptooth, 2020-04-29:
> Have the issues been resolved with r5115?

not tested yet. little bit busy... I will report.

cryptooth, 2020-04-29:
Problem solved with r5115. Thanks.

closed 2020-04-29 by cryptooth

asava, 2020-04-29:
Thanks for confirming.

malhal, 2020-04-30:
Just upgraded successfully from the stable 5108 to 5115 so I can also confirm whatever the problem was has now been fixed.

asava, 2020-04-30:
Thanks for your comment. Glad to hear it's finally working.

vacarogit, 2020-05-14:
> Thanks, still on 5108 waiting for this to be fixed will keep waiting

Same,  on 5108 since March, any report of instability/errors has been banned by author...any workarounds for you?