CloverHackyColor/CloverBootloader #148: clover installation error

danipdj13, 2020-04-28:
i cant install clover bootloader into my usb, any solutions for this?
[image]

talin3, 2020-04-28:
The same issue for me, unfortunately.

danipdj13, 2020-04-28:
It is on virtual box so I can’t use the recovery drive
I’ve seen that it can be fixed using csrutil disable on the recovery, but I can’t do it

vectorsigma72, 2020-04-28:
Hi guys, no one tested it on a virtual machine but I'll be happy to help you. Please try attached app and let me 
[Clover.app_v1.19.pkg.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4545730/Clover.app_v1.19.pkg.zip)

know.

danipdj13, 2020-04-28:
I tried it out, and the message changed
[image]

vectorsigma72, 2020-04-28:
Thanks, now I discovered that this error reported by someone else comes from a virtual machine, finally. But I'm sorry can't do nothing, unless there is something you can do setting up the VM.

danipdj13, 2020-04-28:
Oh, well, that’s disappointing 
Does anyone have a vmdk or something from a older version of MacOS? I think that it is because Catalina has some kind of protection for the files

vectorsigma72, 2020-04-28:
Your USB device is not affair of Catalina i guess and I'll be surprised if this will work in Mojave and below.

danipdj13, 2020-04-28:
How could I fix it?
I’m downloading a high Sierra version to try it

vectorsigma72, 2020-04-28:
Thanks, but sorry I have no idea right now. At this point I'm not even sure if the volume's mount point is real, it's a sym link or whatever.

danipdj13, 2020-04-28:
Thanks anyway
I close this post

closed 2020-04-28 by danipdj13