CloverHackyColor/CloverBootloader #149: Memory panic

danipdj13, 2020-04-29:
Hello everyone
I managed to make a usb bootable but, I have a problem when I load it up 
[image]
I don’t know what else to do
Thanks in advance

CMMChris, 2020-04-29:
1st: Not a Clover issue
2nd: Nobody can help you without the relevant information which is entirely cut off in your picture.

danipdj13, 2020-04-29:
Seems like it disappeared 
Now when it pass the test, it suddenly stops giving video signal

SergeySlice, 2020-04-29:
Set
		<key>PanicNoKextDump</key>
		<true/>
to see more informative panic.

danipdj13, 2020-04-29:
Now the panic message doesn’t appear, it just stop sending signal to my monitor

asava, 2020-04-29:
Did it work with previous Clover or it is the first time you try?

danipdj13, 2020-04-29:
It’s my first time building a hackintosh

asava, 2020-04-29:
This is not a support forum. Please visit the forums for that.

closed 2020-04-29 by asava