CloverHackyColor/CloverBootloader #151: Apple Progress Load bar issue

mehmetyukcell, 2020-04-29:
Clover bootloader issiue: Apple progress load bar not appeared when clover 5107 from to 5115. But, in the clover 5107 is not a problem.

asava, 2020-04-29:
You have by mistake CustomLogo set anywhere in your config.plist?

mehmetyukcell, 2020-04-29:
> You have by mistake CustomLogo anywhere in your config.plist (set to anything other than false)?

Sory, I beg your pardon! CustomLogoTheme
[image]

mehmetyukcell, 2020-04-29:
> You have by mistake CustomLogo anywhere in your config.plist (set to anything other than false)?

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4555140/config.plist.zip)

asava, 2020-04-29:
So the problem is solved by setting to false?

closed 2020-04-29 by asava

mehmetyukcell, 2020-04-29:
> So the problem is solved by setting to false?

which setting should I choose? 
[image]

asava, 2020-04-29:
None or No