CloverHackyColor/CloverBootloader #152: Clover r5115 - Audio Not Working

najeebanwer, 2020-05-01:
I have macOS High Sierra (10.13.6) installed on a Dell XPS M1340. The specs are:
CPU: 2.67 GHz Intel Core 2 Duo
RAM: 8 GB 1333 MHz DDR3
GPU: NVidia GeForce 9400M G 256 MB

I have been using Clover r5114 and everything appeared to be working correctly, and has been working for years through many Clover revisions. I upgraded to Clover r5115 and the audio stopped working.

I have a modified AppleHDA.kext with resource files for layout12.xml, Platforms.xml and a helper kext called HDAIDT.kext that injects the codec id/layout. All of the details of the audio kexts are in the attached file.

I have also included all the relevant information (IORegistryExplorer file, bootlog etc.) for both Clover r5114 and r5115. Please let me know if you need any other information to help with this issue. Thanks.
[Clover r5115 Audio Issue.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4562720/Clover.r5115.Audio.Issue.zip)

asava, 2020-05-01:
Please post your config.plist, or at least the section relevant to audio.

najeebanwer, 2020-05-01:
Attached is my config.plist. I have only redacted some information from the SMBIOS key.

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4565336/config.plist.zip)
Thanks for your help.

asava, 2020-05-01:
Unfortunately, I cannot easily identify the problem.
Do you have the possibility to build clover? If yes, we would probably need you to help us identify which is the commit which breaks it, as It could be many things.

Specifically, it would be good to start with these commit:
start by testing 5eef6624f11cdb4c91d8f80237f118fcaf0dc159
if it is doesn't work, test 8d35aa3ecc2dd682225937bd48b3330c8f58ce98
Judging on these results, we can go on.

najeebanwer, 2020-05-01:
Sorry, I'm a newbie when it comes to building Clover. Do I need to clone the repository first? Is there any setup required (environment variables, etc.) for the git command to work? This is what I get:
[image]

Would it be possible for you to provide the required builds so that I can test them? I don't want any mistakes I make in the build to affect the results.

asava, 2020-05-01:
I could, but I don't know how many we'll have to test, so it's probably more convenient if you build yourself. And It's very simple really:
to clone: 
mkdir ~/src; cd ~/src
git clone https://github.com/CloverHackyColor/CloverBootloader.git

to choose the commit you wish:
git checkout 5eef662

to build:
./buildme, choose option 1. first time you build will take a rather long time as the toolset is downloaded and built. then on next tests it will be faster.

When done, copy CloverPackage/CloverV2/EFI/CLOVER/CLOVERX64.efi to your EFI directory, overwriting the old one (or BOOTX64.efi if you are using that instead).

Then, reboot, you should see reported revision 5114 in Clover. If you see still 5115, you probably didn't copy right.

najeebanwer, 2020-05-01:
Okay, I get it now. This is what I see:
[image]
I will get back to you with the results.

asava, 2020-05-01:
The problem is that there were many changes between 5114 and 5115, so as I cannot reproduce it,  we need to narrow down. The idea is just to find which commit breaks it. Once we find, I will inspect the changed code in that commit, and hopefully see what's wrong and fix it.

najeebanwer, 2020-05-02:
Still trying to build commit 5eef662. This is the error I get:
[image]

najeebanwer, 2020-05-02:
I was not able to build commit 5eef662.
I was able to build commit 8d35aa3 (I made sure the correct version was booted by checking the Clover about screen) and the audio still works in that version.

asava, 2020-05-02:
Thank you. We have a start.
Sorry, it was my mistake, because 5eef662 really doesn't build.
Please try b609bfef1ce9e1e1cace4e84cc92420f99f07017

najeebanwer, 2020-05-02:
Commit b609bfe: Audio does NOT work.
Please let me know which one to try next.

asava, 2020-05-02:
Alright, I see a potential problem in between these two commits.
Please test this build: [this build](https://wetransfer.com/downloads/e14aea3118ed9badad2e9170432930bd20200502220936/a5aecd923537bdbe59dc086ad3a9b7a120200502220952/d688d1)
And confirm whether the problem continues or is solved.

najeebanwer, 2020-05-02:
The build you provided (variant of b609bfe): Problem solved, audio works.

asava, 2020-05-03:
Good. Thank you for your help in determining the issue. I committed the fixes to master. You can compile current version if you wish and it should work:
git checkout master
git pull
./buildme

najeebanwer, 2020-05-03:
Thanks for your quick response and for getting this fixed. 
You provided very clear instructions for building and testing Clover.
Unfortunately, I am unable to build the latest master; I get the following errors:
[image]
But that's okay; I can wait for the next stable release.

As an aside, I would suggest that someone on the team update the Clover Wiki (https://github.com/CloverHackyColor/CloverBootloader/wiki/Building) with the instructions that you provided.

closed 2020-05-03 by najeebanwer

asava, 2020-05-03:
Not sure why you are getting that error, I just tried and it built successfully.
Seems like you have some missing library, but I don't understand why. Did you cd to the directory before running buildme? If not, maybe you should try to start from scratch:
```
mkdir ~/src; cd ~/src
git clone https://github.com/CloverHackyColor/CloverBootloader.git
cd CloverBootloader
./buildme, choose option 1.

najeebanwer, 2020-05-03:
Okay, that worked. I was able to build the latest master and the audio still works.
[image]

asava, 2020-05-04:
Thank you for confirming.

asava, 2020-05-06:
Confirmed as resolved in 5116.

najeebanwer, 2020-05-06:
Agreed. I installed Clover r5116 and the audio still works. Thanks.