CloverHackyColor/CloverBootloader #153: Failed build of clover on Ubuntu 18.04 LTS

darrenjkt, 2020-05-02:


asava, 2020-05-02:
did you try to use ./buildme script?

darrenjkt, 2020-05-02:
I just tried it with the ./buildme script with option 7 (build all) and it also failed. I've posted the output below. 

```
TOOLCHAIN: GCC53 (override example: './buildme XCODE8')

 1) build Clover
 2) build Clover with HFSPlus
 3) make pkg
 4) make app
 5) make app (with Clover)
 6) make iso
 7) build all
 8) test build (no autogen, no boot files)
 9) status
10) update Clover
11) show diff
12) open drivers directory
13) update Clover (reset changes)
14) clean BaseTools
15) quit
Please enter your choice: 7
WORKSPACE: /home/darren/Desktop/CloverBootloader
EDK_TOOLS_PATH: /home/darren/Desktop/CloverBootloader/BaseTools
CONF_PATH: /home/darren/Desktop/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/darren/Desktop/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/darren/Desktop/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/darren/Desktop/CloverBootloader/Conf/target.txt
[BUILD CLOVER]
TOOLCHAIN_DIR: /usr
NASM_PREFIX: /usr/bin/
NASM_VER: 2.13.02
Initializing workspace
WORKSPACE: /home/darren/Desktop/CloverBootloader
EDK_TOOLS_PATH: /home/darren/Desktop/CloverBootloader/BaseTools
CONF_PATH: /home/darren/Desktop/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/darren/Desktop/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/darren/Desktop/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/darren/Desktop/CloverBootloader/Conf/target.txt

Running edk2 build for CloverX64 using the command:
build  -D DISABLE_USB_SUPPORT -D NO_GRUB_DRIVERS_EMBEDDED --conf=/home/darren/Desktop/CloverBootloader/Conf -D USE_BIOS_BLOCKIO -D USE_LOW_EBDA -p Clover.dsc  -a X64 -b RELEASE -t GCC53 -n 9 

Build environment: Linux-5.3.0-28-generic-x86_64-with-Ubuntu-18.04-bionic
Build start time: 18:28:28, May.02 2020

WORKSPACE        = /home/darren/Desktop/CloverBootloader
EDK_TOOLS_PATH   = /home/darren/Desktop/CloverBootloader/BaseTools
CONF_PATH        = /home/darren/Desktop/CloverBootloader/Conf

Architecture(s)  = X64
Build target     = RELEASE

Processing meta-data .Toolchain        = GCC53

Active Platform          = /home/darren/Desktop/CloverBootloader/Clover.dsc
..build: : warning: Module MetaFile [Sources] is missing local header!
    Local Header: /home/darren/desktop/cloverbootloader/refit_uefi/cpp_unit_test/xtoolscommon_test.h not found in /home/darren/Desktop/CloverBootloader/rEFIt_UEFI/refit.inf
build: : warning: Module MetaFile [Sources] is missing local header!
    Local Header: /home/darren/desktop/cloverbootloader/refit_uefi/cpp_unit_test/xtoolscommon_test.h not found in /home/darren/Desktop/CloverBootloader/rEFIt_UEFI/refit.inf
 done!
Building ... /home/darren/Desktop/CloverBootloader/Library/MemLogLibDefault/MemLogLibDefault.inf [X64]
Building ... /home/darren/Desktop/CloverBootloader/MdePkg/Library/BaseReportStatusCodeLibNull/BaseReportStatusCodeLibNull.inf [X64]
Building ... /home/darren/Desktop/CloverBootloader/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint.inf [X64]
Building ... /home/darren/Desktop/CloverBootloader/MdePkg/Library/UefiLib/UefiLib.inf [X64]
[... 45 lines cut ...]
[CC] ARShiftU64
[CC] IoLibGcc
[CC] UefiDriverModel
[CC] CopyMemWrapper
[SLINK] BasePcdLibNull
Building ... /home/darren/Desktop/CloverBootloader/Library/OcGuardLib/OcGuardLib.inf [X64]
[CC] DevicePathToText
[CC] IoLibMmioBuffer
cc1: all warnings being treated as errors
[CC] CxxSymbols
GNUmakefile:360: recipe for target '/home/darren/Desktop/CloverBootloader/Build/Clover/RELEASE_GCC53/X64/Library/MemLogLibDefault/MemLogLibDefault/OUTPUT/printf_lite.obj' failed
make: *** [/home/darren/Desktop/CloverBootloader/Build/Clover/RELEASE_GCC53/X64/Library/MemLogLibDefault/MemLogLibDefault/OUTPUT/printf_lite.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/darren/Desktop/CloverBootloader/Build/Clover/RELEASE_GCC53/X64/Library/MemLogLibDefault/MemLogLibDefault]

build.py...
 : error F002: Failed to build module
	/home/darren/Desktop/CloverBootloader/Library/MemLogLibDefault/MemLogLibDefault.inf [X64, GCC53, RELEASE]

- Failed -
Build end time: 18:28:34, May.02 2020
Build total time: 00:00:06
```

asava, 2020-05-02:
@jief666 The error reported is at printf_lite

jief666, 2020-05-03:
Let me check.

asava, 2020-05-03:
The reported error:
`CloverBootloader/Library/MemLogLibDefault/printf_lite.c:532:1: error: 'no_sanitize' attribute directive ignored [-Werror=attributes]
 static void print_longlong(INT_BIGGEST_TYPE v, unsigned int base, PrintfParams* printfParams) __attribute__((no_sanitize("undefined"))); // disable sanitize because we'll do -v`

Strangely I don't get it on Ubuntu 20.04

jief666, 2020-05-03:
I don't have that problem either. I guess it's a GCC version problem. 

@darrenjkt 
What GCC version do you use ?
Anyway, so far this attribute is needed only for Xcode. So I've committed a fix. Could you check and close ? Thanks.

asava, 2020-05-05:
Did you check if this issue has been resolved?

jief666, 2020-05-05:
I've corrected the #ifdef, so "__attribute__((no_sanitize("undefined")))" only happens for clang (no real need for gcc yet, this is for disabling a warning is Xcode).
So @darrenjkt should not have the problem anymore. But I was still curious why he has that problem when compiling with GCC and not me. I'd like to know which GCC version he uses.
I guess we can close. @darrenjkt can still answer if he wants to in a closed issue, I think.

darrenjkt, 2020-05-06:
Sorry for the late response. My gcc version is 7.5.0. Due to some issues with my laptop I won't be able to check if this works anytime soon, my apologies. However thank you so much for your help, I shall close the issue.

closed 2020-05-06 by darrenjkt