CloverHackyColor/CloverBootloader #155: Clover EFI Shell Icon

ouija, 2020-05-02:
Have a strange issue occuring after upgrading to latest version of Clover and the "Boot EFI Shell" icon being huge in size in relation to all other icons.   Any ideas what might be causing this?

Thanks!

asava, 2020-05-02:
Most probably a problem with your theme.
Which theme you are using? Also a link to it, to see if it can be reproduced.

ouija, 2020-05-03:
I'm using a modified version of the "bluemac" theme, but the original does this as well:

[Imgur](https://imgur.com/eWclLLa)

Looks like it is caused by the theme having both a tool_shell.icns file and tool_shell.png, with clover loading in the .icns file _(removing or renaming this fixes the problem)_

So likely not an issue with Clover but this theme itself.   Also noticed that Clover is now properly referencing the "os_unknown.icns" file for this Android partition I have, which used to reference "os_clover.icns" -- Maybe a request to add/enable support for detecting Android_x86 partitions?

Consider this issue resolved :)

closed 2020-05-03 by ouija

asava, 2020-05-03:
I took a look at that theme, and yes, the .icns (which is being attempted to load first) was of wrong size, if it was or right size it wouldn't have mattered. So it was just a theme issue. I'm not sure what's the relation to Nvidia machine.

Just for the case I consider looking at detecting Android-x86 machines, send a screenshot after you press space on the "unknown" partition.
You are compiling and using Clover on a x86-android?