CloverHackyColor/CloverBootloader #156: Gigabyte z390 boot compatible issue

Raydem0n, 2020-05-04:
On my hackintosh with Gigabyte z390, it requires BOOTX64.efi with clover "508x" version to run into Clover menu, and then it requires BOOTX64.efi with clover "5107" version to run into MacOS X.

I have two hard disks, one is installed with OS X 10.15.1, and another 10.15.4. If I boot from 10.15.4 at first, it would stubbed at BIOS logo screen. My working operations are here,
1. Boot from OS X 10.15.1 hard disks.
2. In OS X 10.15.1, do reboot
3. Boot from OS X 10.15.4 hard disks, and it run into clover menu.
4. Choose 10.15.4 hard disks and boot into OS X 10.15.4

Even clover 5115 version contains the issue. Hope the next version will fix it.

Raydem0n, 2020-05-04:
Attached my CFG Lock info,

MSR_PMG_CST_CONFIG_CONTROL........(0xE2) : 0x1E008000
------------------------------------------
 - I/O MWAIT Redirection Enable......... : 0 (not enabled)
 - CFG Lock............................. : 1 (MSR locked until next reset)
 - C3 State Auto Demotion............... : 1 (enabled)
 - C1 State Auto Demotion............... : 1 (enabled)
 - C3 State Undemotion.................. : 1 (enabled)
 - C1 State Undemotion.................. : 1 (enabled)
 - Package C-State Auto Demotion........ : 0 (disabled/unsupported)
 - Package C-State Undemotion........... : 0 (disabled/unsupported)

Raydem0n, 2020-05-04:
The same problem remains with CFG Lock disabled.
After disable CFG Lock by grub setup_var_3,
the CFG status are shown below,

MSR_PMG_CST_CONFIG_CONTROL........(0xE2) : 0x1E000000
------------------------------------------
 - I/O MWAIT Redirection Enable......... : 0 (not enabled)
 - CFG Lock............................. : 0 (MSR not locked)
 - C3 State Auto Demotion............... : 1 (enabled)
 - C1 State Auto Demotion............... : 1 (enabled)
 - C3 State Undemotion.................. : 1 (enabled)
 - C1 State Undemotion.................. : 1 (enabled)
 - Package C-State Auto Demotion........ : 0 (disabled/unsupported)
 - Package C-State Undemotion........... : 0 (disabled/unsupported)

Raydem0n, 2020-05-04:
After disable CFG Lock by grub setup_var_3, and replace clover 5115 with 5093, then the problem fixed. Guess the compatible issue is still produced by CFG Lock.

asava, 2020-05-04:
I am sorry, but I understand nothing.
You talk about 3 different Clover versions (508x, 5107, 5115) 
Two different macOS versions (10.15.1, 10.15.4)
But I still don't understand what you are trying to do.

asava, 2020-05-05:
If no further comments, I will close this issue.

closed 2020-05-05 by asava