CloverHackyColor/CloverBootloader #157: The menu text cannot be displayed completely

allegrox, 2020-05-04:
How to increase the effective length of menu item text?
And in the interface of attachment 2, the keyboard response becomes very slow and the CPU fan accelerates obviously, but it is not the case in other menu items

[image]
[image]

asava, 2020-05-04:
Seems like a bug. Please specify which theme you are using, attach your theme, and I will take a look.

allegrox, 2020-05-04:
Yes, I am using r5115.
And my theme in attachment named mac_osx.zip.
[image]
[mac_osx.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4573827/mac_osx.zip)

asava, 2020-05-04:
Yes, text width was a bug. Fixed in master, if you build from sources it will be OK.
Regarding CPU usage, it is the same procedure used for all menus.
Maybe because that menu doesn't have Anime you see some difference.

asava, 2020-05-05:
Further testing with your theme, I found some more bugs related to proportional fonts and fixed them, some really old bugs.

For example, I saw that you could not edit some items in options menu.
Or, that you had to set in theme a very large font size because of a bug, resulting in hardly usable options menu.
I fixed them too, so now I can recommend you to change in your theme.plist CharWidth=14, to get a much better experience with Options menu which is now possible.

Once you confirm everything is fine, feel free to close this issue.

allegrox, 2020-05-05:
I tried to compile the clover bootloader and stopped when I encountered the following error. How can I fix this error?
`Initializing workspace
WORKSPACE: /Users/xiyue/src/CloverBootloader
EDK_TOOLS_PATH: /Users/xiyue/src/CloverBootloader/BaseTools
CONF_PATH: /Users/xiyue/src/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /Users/xiyue/src/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /Users/xiyue/src/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /Users/xiyue/src/CloverBootloader/Conf/target.txt
Building tools as they are not found
make -C Source/C
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
mkdir ./libs
make -C Common
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  BasePeCoff.c -o BasePeCoff.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  BinderFuncs.c -o BinderFuncs.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  CommonLib.c -o CommonLib.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  Crc32.c -o Crc32.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  Decompress.c -o Decompress.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  EfiCompress.c -o EfiCompress.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  EfiUtilityMsgs.c -o EfiUtilityMsgs.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  FirmwareVolumeBuffer.c -o FirmwareVolumeBuffer.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  FvLib.c -o FvLib.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  MemoryFile.c -o MemoryFile.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  MyAlloc.c -o MyAlloc.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  OsPath.c -o OsPath.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  ParseGuidedSectionTools.c -o ParseGuidedSectionTools.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  ParseInf.c -o ParseInf.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  PeCoffLoaderEx.c -o PeCoffLoaderEx.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  SimpleFileParsing.c -o  [... cut ...] ueCommon.o
mkdir ./bin
make -C VfrCompile VfrLexer.h
BIN_DIR='.' make -C Pccts/dlg
cc -O -I. -I../support/set -I../h -DUSER_ZZSYN -DZZLEXBUFSIZE=65536 -c dlg_p.c
In file included from dlg_p.c:14:
In file included from ../h/pcctscfg.h:61:
../h/pccts_stdio.h:7:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
make[3]: *** [dlg_p.o] Error 1
make[2]: *** [Pccts/dlg/dlg] Error 2
make[1]: *** [VfrCompile/VfrLexer.h] Error 2
make: *** [Source/C] Error 2`

asava, 2020-05-05:
Not sure how you are trying to compile, but something is not right.
To compile:
```
mkdir ~/src; cd ~/src
git clone https://github.com/CloverHackyColor/CloverBootloader.git
cd CloverBootloader
./buildme
```
and choose option 1

allegrox, 2020-05-05:
Yes, my steps are the same as yours. 
I found more than one stdio.h on the disk, but I don't know which to refer to?

asava, 2020-05-05:
Do you have Xcode installed? If yes, start it and in Xcode -> Preferences -> Command line Tools, make sure it is set correctly to last one.

asava, 2020-05-05:
In any case, [here is a binary](https://we.tl/t-q2gdNufI2m).
Overwrite your CLOVERX64.efi (and BOOTX64.efi if needed) with the file included.

allegrox, 2020-05-05:
Thank you very much, I have found the error. Although I installed Xcode and Command line Tools, But it was not checked in the Xcode -> Preferences. 
It has now been compiled normally.

closed 2020-05-05 by allegrox

asava, 2020-05-06:
Confirmed as resolved in 5116.