CloverHackyColor/CloverBootloader #158: "Cannot install without ensuring the file system is read-write" when trying to install on a USB drive

jonasrk, 2020-05-04:
Hi friends,

I'm on a MacBook and want to use r5115 to install on a USB Installer drive. It fails with "Cannot install without ensuring the file system is read-write" I'm assuming because it can't get full access to my MacBooks drive.

It should only need full access to the USB drive, which it has, or am I missing something?

asava, 2020-05-05:
@vectorsigma72

asava, 2020-05-05:
Duplicate of [https://github.com/CloverHackyColor/CloverBootloader/issues/22](url) ?

jonasrk, 2020-05-05:
@asava I read through #22 and #6 entirely, however I think it's not a duplicate. Those two issues were mainly about updating on a Hacky.

My issue here is about creating a first USB installation drive on Mac hardware. This is currently not possible for me, because of the crash described above. Since all we want here is to flash the USB drive, this should work without disabling SIP or getting full root access to my Mac SSD.

Please also note the comment https://github.com/CloverHackyColor/CloverBootloader/issues/22#issuecomment-613796467 and the number of upvotes it received.

@asava , can you remove the duplicate label so we can start working on fixing this issue? :)

vectorsigma72, 2020-05-05:
> @vectorsigma72

Hi guys, this should tagged as invalid (as I already tried to explain why multiple times) and i'm going to explain why again: 

1. In Catalina the root file system becames read write.
2. The package is just an archive and not a program (as always).
3. A package is handled by /System/Library/CoreServices/Installer.app, and we can't do nothing. 
4. Installer.app knows when you are potentially trying to install to /, ready to deny anything will desecrate your Mac.
5. Clover package can potentially install somethings to / (preferences, rc scripts, the pref panel and/or just the EFI directory).
6. In Catalina Installer.app knows 4. and 5. and prevent the installer to run it,  even if you want to install on a USB pen drive (no matter what). The workaround is to make the root filesystemr ead/write before choosing the target volume and allow any volume to be chosen.

If you don't want to make the root filesystem read/write, in any cases... there is no way to continue the installation. No matter if you run from a precious vanilla Mac even if you want to install to an external drive: the package has no brain, and you will have no chances... conscious or not.

To avoid this kind of issue I created Clover. app which can mount ESP partitions without creating a fake volume mount points (like the package do with the sysmlink to the ESP) and which doesn't install anything to the root file system. Use it. 
Alternatives: continue to open issues on github :-) without chances.

vectorsigma72, 2020-05-05:
> @vectorsigma72

Hi guys, this should be tagged as invalid (as I already tried to explain why multiple times) and i'm going to explain why again:

In Catalina (and newer) the root file system becomes read only.
The package is just an archive and not a program (as always).
A package is handled by /System/Library/CoreServices/Installer.app, and we can't do nothing.
Installer.app knows when you are potentially trying to install to /, ready to deny anything will desecrate your Mac (or your Hack).
Clover package can potentially install somethings to / (like preferences, rc scripts, the pref panel and/or just the EFI directory), even if your will is to install to another drive.
In Catalina Installer.app knows 1., 4. and 5. and prevent the installer to run it, even if you want to install on a USB pen drive (no matter what). The workaround is to make the root filesystem read/write before chosing the target volume and allow any volume to be chosen, and is what the package does with a java script when it ask if you want to make "/" read write.
If you don't want to make the root filesystem read/write (which is temporary until next reboot, at least in Catalina (don't know for future OSes)), in any cases, there is no way to continue the installation. No matter if you run from a precious vanilla Mac, even if you want to install to an external drive: the package has no brain, and you will have no chances... conscious or not.

To avoid this kind of issue I created Clover.app which can mount ESP partitions without creating a fake volume mount point (like the package do with the sysmlink to the ESP) and which doesn't install anything to protected paths. Use it.
Alternatives: continue to open issues on github :-) without chances.

We should left this issue always opened to prevent again users open newer ones.

vectorsigma72, 2020-05-05:
> @vectorsigma72

Hi guys, this should be tagged as invalid (as I already tried to explain why multiple times) and i'm going to explain why again:

1. In Catalina (and newer) the root file system becomes read only.
2. The package is just an archive and not a program (as always).
3. A package is handled by /System/Library/CoreServices/Installer.app, and we can't do nothing.
4. Installer.app knows when you are potentially trying to install to /, ready to deny anything will desecrate your Mac (or your Hack).
5. Clover package can potentially install somethings to / (like preferences, rc scripts, the pref panel and/or just the EFI directory), even if your will is to install to another drive.
6. In Catalina Installer.app knows 1., 4. and 5. and prevent the installer to run it, even if you want to install on a USB pen drive (no matter what). The workaround is to make the root filesystem read/write before chosing the target volume and allow any volume to be chosen, and is what the package does with a java script when it ask if you want to make "/" read write.

If you don't want to make the root filesystem read/write (which is temporary until next reboot, at least in Catalina (don't know for future OSes)), in any cases, there is no way to continue the installation. No matter if you run from a precious vanilla Mac, even if you want to install to an external drive: the package has no brain, and you will have no chances... conscious or not.

To avoid this kind of issue I created Clover.app which can mount ESP partitions without creating a fake volume mount point (like the package do with the sysmlink to the ESP) and which doesn't install anything to protected paths. Use it.
Alternatives: continue to open issues on github :-) without chances.

We should left this issue always opened to prevent again users open newer ones.

dokuruharsha, 2020-05-17:
HI ,

I am new here so please bear with me 

I am having the same issue @jonasrk was having

To give a background, I am trying to build my 1st hackintosh and I have ran into the issue. 

I am using Macbook pro with MacOS Catalina 10.15.4. I have downloaded and copied OS files to my flash drive. When i tried to download and install Clover installer (Clover_v2.5k_r5103.pkg
) i am seeing the error "Cannot install without ensuring the file system is read-write". 

After reading the above comments by @vectorsigma72 and related Issues (#22 and #6 ), I understand that MacOS Catalina have read only access to root filesystem inorder to help prevent the accidental overwriting of critical operating system files and requires write access. The steps are not very clear for me in the above comments, could anyone please provide clear steps, so that i can install clover on my flashdrive?

PS : I am not a geek and new to apple ecosystem. 

Really appreciate any comments

Thanks
Harsha

dokuruharsha, 2020-05-17:
HI All,

I have revisited the comments on #6 . And the below comment from @gomsoup worked like a charm

1. restart your mac
2. press command + r until appear Apple's logo
3. Select your account and enter password
4. go to menubar-utility-terminal
5. type this command  - " csrutil disable " 

However my issue is now resolved, for information purposes i would like to know if this is the only way to give edit access to rootfile system ?

Any comments are appreciated

Thanks
Harsha

vectorsigma72, 2020-05-20:
Hi @dokuruharsha, maybe you lost the "Clover.app” part?

dokuruharsha, 2020-05-20:
Hi @vectorsigma72 ,

Thank you for taking time to reply my query. That is helpful.

I have installed Clover configurator and downloaded the bootloader Package to make my MacOSX bootable drive sucessfully

rcknr, 2020-07-31:
I used the following workaround:

1. Download Clover bootloader package (e.g. Clover_r5120.pkg)
2. Extract the package: `pkgutil --expand Clover_r5120.pkg CloverPkg`
3. Edit `CloverPkg/Distribution`: find `checkFileSystemIsWritable()` function and add `return writable;` after `var writable = true;`
4. Create new package with modified file: `pkgutil --flatten CloverPkg Clover_r5120_mod.pkg`
5. Install the modified package.

closed 2021-04-11 by SergeySlice