CloverHackyColor/CloverBootloader #159: Ubuntu custom menuentry

florindogarini, 2020-05-05:
Hi, with Clover r5115 Ubuntu 20.04 custom menuentry has disappeared. I can login with the F3 key. Windows 10 and Catalina custom menuentry work correctly.
Thanks for your work.

asava, 2020-05-05:
please provide details on how your custom entry is set up, the relevant part from config.plist

florindogarini, 2020-05-05:
</dict>
	<key>GUI</key>
	<dict>
		<key>Custom</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>FullTitle</key>
					<string>Windows</string>
					<key>Ignore</key>
					<false/>
					<key>Type</key>
					<string>Windows</string>
					<key>Volume</key>
					<string>87940379-2245-4F3C-BE10-40735E4585FF</string>
					<key>VolumeType</key>
					<string>Internal</string>
				</dict>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>FullTitle</key>
					<string>Catalina</string>
					<key>Ignore</key>
					<false/>
					<key>NoCaches</key>
					<false/>
					<key>Type</key>
					<string>OSX</string>
					<key>Volume</key>
					<string>Catalina</string>
					<key>VolumeType</key>
					<string>Internal</string>
				</dict>
				<dict>
					<key>Disabled</key>
					<false/>
					<key>FullTitle</key>
					<string>Ubuntu</string>
					<key>Ignore</key>
					<false/>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>46556EFF-15C2-40A5-A3C1-36ED32D71EB4</string>
					<key>VolumeType</key>
					<string>Internal</string>
				</dict>
			</array>
		</dict>
		<key>Hide</key>
		<array>
			<string>DATA I</string>
			<string>Preboot</string>
			<string>Recovery</string>
			<string>Microsoft</string>
			<string>CatalinaTest</string>
			<string>Ubuntu</string>

asava, 2020-05-05:
That is because Linux path detection seems not to work currently for Linux, due to some other changes. I'll look into it.

asava, 2020-05-05:
Please check if the last commit solves your issue (you need to build from sources).

florindogarini, 2020-05-06:
Hi, I'm sorry, I'm not able to compile. Could you please do it?

asava, 2020-05-06:
[here](https://we.tl/t-emzidpChEJ) is a binary.
Replace your CLOVERX64.efi (and BOOTX64.efi if needed) from your EFI folder with this one.
You can check if you upgraded correctly  in Clover's about screen ('a' from GUI) - you should see there latest commit (8f5a461..)

asava, 2020-05-06:
You can test 5116 as it was released.

florindogarini, 2020-05-07:
Hi, it works! Many thanks for your work.

asava, 2020-05-07:
Thanks for confirming.
Confirmed as fixed in 5116.

closed 2020-05-07 by asava