CloverHackyColor/CloverBootloader #161: Clover r5116 appears r5115

moris49, 2020-05-06:
Hi!
After reinstalling r5116 and replacing my EFI folder in ESP (legacy bios, GUID part with 10.10.5), when I reboot I have a 1st clover boot screen (with shell64 access but no visible partition), the version of this screen is named r5115. I have to select exit and after boot6 I have clover menu with my partitions and clover version r5115 too. 
2 questions:- is it normal that the version is still noted r5115?
- how can I get rid of this double clover boot issue? (which I don't have with USB)?
best regards
Nico

moris49, 2020-05-06:
clover configurator updater indicates that the current boot version is 5115. If I select CLOVERX64.efi in my EFI part and clic update, I can see 'CLOVERX64.efi not found!' in the bottom left corner...

asava, 2020-05-06:
No, this is not normal. If you see version 5115, the update didn't work. I tested the binary in the 5116 release and it shows properly 5116. Which means, that if it shows 5115, you are still using 5115. Sounds like it didn't get installed, or it got installed to to the wrong place.

moris49, 2020-05-06:
Thanks, I updated the issue because I resolved this by reupdating simply. from the package.
Still, I don't know why the updater says CLOVERx64.efi not found.
I updated thinking it could resolve the real issue described (clover booting twice).
regards

asava, 2020-05-06:
If it's resolved, please, close the issue. Also the booting twice, means you have clover in two different places installed.

moris49, 2020-05-06:
Ok I don't know how it is possible. (and I can't find any other location where it could be installed) I changed the title of the issue (refresh the page?) to focus on the double boot. forums don't provide any info on this issue. I have been looking on TM and IM for a while.
If this place is not relevant to try to resolve this (I thought it was part of clover installation), then I will close the issue. 
I already cleaned the EFI partition and reinstalled.
Thanks again

asava, 2020-05-06:
You are saying you are booting Legacy. So it might be also somehow related to a legacy driver.
Did you use legacy clover before in the same conditions, with a previous version, and didn't have this issue?

moris49, 2020-05-06:
I had chimera before, then tested clover on USB (without ESP), and then installed on SSD, and cleaned chimera (erase extra and kexts). I tried reinstalling with boot0ss or boot0af but same.
I have only 1 SSD connected since I use clover (for test purposes).

asava, 2020-05-06:
I believe you have two different legacy firmware somewhere.
Check on all your partitions for /boot* files (in the root directory). Erase them all. Reinstall again. We have two boot variants - probably one works and the other doesn't on your system. You need to find the one that works.

closed 2020-05-06 by asava

moris49, 2020-05-06:
Thanks Asava,
the 1 first clover screen appears before loading boot6 or 7 (I can see the 6 or the 7 when the 2d boot screen is loading). It's installed both for UEFI and Legacy. NVRAM Cleaned before.

asava, 2020-05-06:
So, now that you explain better, I can help better. What this means:
First UEFI version loads, it does not work properly. Then, you exit, Legacy loads and it works.
Meaning you probably have a problem with your partition driver in your UEFI bios.
Add in UEFI Drivers -> Additional drivers -> PartitionDXE. It might solve it.

moris49, 2020-05-06:
Legacy bios here.
I tried the additional drivers but it doesn't change anything.
If I erase the EFI partition content, and reboot, no bootloader.
then I reinstall clover, without replacing any file, I have the same issue.
1st screen EFI is 2.10 , firmware: TianoCore.org rev 4096.0001
2st screen EFI is 2.70, firmware: x86_64 (64bit)
??
When I boot from USB, I have directly the 'good' bootloader

asava, 2020-05-06:
That confirms my first theory, that you have two legacy firmwares. v2.10 is also a legacy firmwares, but not installed by Clover. Did you search all your partitions for boot* files? Maybe one of your partitions has bootcode that loads another firmware from another path, I cannot predict what it is.

moris49, 2020-05-07:
deledeted boot* in /i386 reinstalled clover, reboot, same.
Updated to High Sierra via app store hoping for a cleaning in the filesytem, same. Last scurity update, same...
Still GUID part, converted to apfs, with legacy boot. Don't have any Idea of where it could launched!

asava, 2020-05-07:
What can I say? This is what happens when you use shady tools.
If you don't know how to detect the problem, zap your whole partition table and start from scratch.