CloverHackyColor/CloverBootloader #162: SMBIOS handling

sincohen, 2020-05-08:
feature request:

Can SMBIOS data be separated from config.plist (in separate smbios.plist for instance)? separating this out would make EFI much more (fully?) portable. people could post working EFI or config.plist and SMBIOS data will always be secure, persistent, and easy to add and maintain separately.

vectorsigma72, 2020-05-08:
That make no sense to me. In the event you want to share your configuration there are the exact same chances you forgot to delete your secreted smbios.plist, and otherwise, everyone will wonder which emulation to use. Just don't share the BoardSerialNumber and the MLB (which is under RtVariables).

SergeySlice, 2020-05-11:
No sense to share config.plist. There are patches for kexts which can be shared somewhere in common place. There are ACPI patches which is BIOS version dependent so will be different for each individual. There are GUI tuning which is individual and not obligatory for success.

commit f5d8d1d3 (2025-04-01): fix #162: heap-buffer-overflow in fetch_interval_quantifier due to double PFETCH
closed 2020-05-11 by SergeySlice