CloverHackyColor/CloverBootloader #163: Please help my fix error Battery percentage is not displayed on status bar hackintosh when update clover boot higher version 5107

phamngocson2020, 2020-05-11:
Hi Clover Bootloader team dev !
My name's Pham Ngoc Son, i'm  35 years old.
I'm very happily  when Clover boot to install Macos on my laptop ( Dell Vostro 3468). Thanks team alot.
But  Clover boot update versions higher 5107 then My laptop don't have % battery on status bar macOs  ( ACPIBaterrymanagent.kext already install).
My laptop ( dell vostro 3468) and other friends too.
I hope  development team will help fix the above error for future updates !!
Thanks CloverBootloader team so much.

Pham Ngoc Son

vectorsigma72, 2020-05-11:
This depend from your DSDT, isn't? My laptop show it  without problems... you tried to go in Sysyem Preferences->Energy Saver->Show battery status in menu bar??

SergeySlice, 2020-05-11:
There is not Clover mistake as I have battery status on me laptop.
And can't be for a common sense.

closed 2020-05-11 by SergeySlice