CloverHackyColor/CloverBootloader #164: Remove "Powered by Clover revision:" from apple_rom_info

BbIKTOP, 2020-05-13:
Is it possible to remove this from the ROM version? It's not good when one asks for support for any program and vendor will se it's not a real mac and deny to help then. Is there any parameter to disable it?

CMMChris, 2020-05-13:
Indeed that would be a good addition to add an option for removing this string.
Anyhow, developers not supporting their products on a hackintosh... is that still a thing? Didn't happen to me in a long time.

BbIKTOP, 2020-05-13:
Right now i have an opened report on parallels about some gui problems appeared with the last update. They asked me to share a "tech report", and the only (ok, first) thing they could understand it's not a real mac, is this string.

BbIKTOP, 2020-05-13:
It is in smbios.cpp, line 1108 and 1109.

BbIKTOP, 2020-05-14:
Hey, ppl, shall I make a pull request for that? What do you think about adding such an option, for example smbios - oemstring? No key means, for example, "keep oem string as it is", empty string means "leave it empty" any other means "insert this value instead" Maybe with macros, like $e for efi version, $r for rom version, $b for board number, $c for clover revision etc?

SergeySlice, 2020-05-18:
There is another issue that we still not resolved: why real mac having the same table doesn't show rom_info vs hackintosh? Something we missed here.
For this reason I will keep this issue open.

SergeySlice, 2020-05-18:
Anyway this settings exists
~~~
	<key>SMBIOS</key>
	<dict>
		<key>NoRomInfo</key>
		<true/>

~~~

closed 2021-05-24 by jief666