CloverHackyColor/CloverBootloader #165: Nothing works since 5109

vacarogit, 2020-05-14:
I tried, I really patiently tried most updates since 5108, that’s the last working version.

I’m fed up by these dangerous updates, none of them work without various problems.
For most serious is 5117 released yesterday, literally won’t boot, send me directly into BIOS...wtf?

Please let go this project and let someone more capable to to the work !!

BbIKTOP, 2020-05-14:
You have to seek for assistance on the appropriate forums, I suppose. What are your specs and what is in the log? Did you try verbose boot mode? Not enough info. Latest clover works just as it is supposed to. I have had precompiled 5117 installed and working, yesterday recompiled it and it still works

vectorsigma72, 2020-05-14:
> I tried, I really patiently tried most updates since 5108, that’s the last working version and for no exaggeration!
> 
> I’m fed up by these dangerous updates, none of them works without various problems.
> For most serious is 5117 released yesterday, literally won’t boot, send me directly into BIOS...wtf?
> 
> Please let go this project and let someone more capable to get the work done !! We appreciate your previous efforts though.
As this was tested and reporting working in the relative forum from the efforts of develepoers and testers and no one of them never see you with a little help (I'm wrong? ), sincerely I can't image other than your personal errors due to any lack of knoledge. If you want an advice buy an iMac and left thinghs that require at least a minumun use of the brain because it's not for you. "patiently” go to boring someone else as unlike what you think, nothing is due to you.

vectorsigma72, 2020-05-14:
> I tried, I really patiently tried most updates since 5108, that’s the last working version and for no exaggeration!
>
> I’m fed up by these dangerous updates, none of them works without various [...]

As this was tested and reporting working in the relative forum from the efforts of devs and testers and no one of them never see you with a little help (I'm wrong? ), sincerely I can't image other than your personal errors due to any lack of knoledge. If you want an advice buy an iMac and left thinghs that require at least a minumun use of the brain because it's not for you. "patiently” go to boring someone else as unlike what you think, nothing is due to you.

closed 2020-05-14 by vectorsigma72