CloverHackyColor/CloverBootloader #167: Clover error

Emanuele-1998, 2020-05-18:
Hi, I have a big problem. always use this efi for opencore and it has always worked but since i formatted it doesn't start anymore. same thing with clover. I tried to format the usb but nothing. 
[image]

what kind of mistake is this?

vectorsigma72, 2020-05-19:
Look at your bootArgs... something is clearly wrong. Press F11 at Clover GUI as I guess them are stored in your nvram, unless your config.plist contains such garbage.

Emanuele-1998, 2020-05-19:
> Look at your bootArgs... something is clearly wrong. Press F11 at Clover GUI as I guess them are stored in your nvram, unless your config.plist contains such garbage.

it doesn't contain anything wrong I believe -v dart = 0 -wegnoegpu -igfxvesa. Anyway I solved updating clover but I don't understand why updating clover it solved if a week ago with that previous version it worked

vectorsigma72, 2020-05-19:
In your screenshot, as I said, I can see something unusual under boot arguments (line 12). But if stopped working with both Clover and opencore as you said... not sure why should be a Clover error. You pressed F11 or not?

Emanuele-1998, 2020-05-20:
> In your screenshot, as I said, I can see something unusual under boot arguments (line 12). But if stopped working with both Clover and opencore as you said... not sure why should be a Clover error. You pressed F11 or not?

Yes I press F11 but don’t work but a week ago with 5103 it worked and went back to work with the last version

vectorsigma72, 2020-05-20:
ok, not taking into account the various prereleases versions of Clover [here ](https://github.com/CloverHackyColor/CloverBootloader/releases) between 5103 and 5117, and considering you cleaned the nvram (what F11 does) as suggested and now working with newer code, I'm going to close this issue as resolved.

closed 2020-05-20 by vectorsigma72