CloverHackyColor/CloverBootloader #168: CloverV2-5118.zip is a zero length file?

exquirentibus, 2020-05-21:
I've downloaded it several times, always a zero length file.  All the others from that release download fine.  Thanks for all your excellent code!!!!

vectorsigma72, 2020-05-21:
Downloading the V2 directory  is ok here.. problem is from your side?

exquirentibus, 2020-05-21:
Well, I downloaded CloverV2-5111.zip as 9.05MB just now - so no problems there - using 
Firefox on Win10.

When I download CloverV2-5118.zip, both Firefox and Chrome show it as 'This file is not commonly downloaded and may be dangerous.'

I notice the link is https://github.com/CloverHackyColor/CloverBootloader/releases/download/5118/CloverV2-5118.zip

but Chrome shows it as:
https://github-production-release-asset-2e65be.s3.amazonaws.com/205810121/cbecf780-9b38-11ea-92ad-2f020f0a8467?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200521%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200521T181343Z&X-Amz-Expires=300&X-Amz-Signature=823b037e1dee5

in their 'downloads' tab with a red warning.

I wasn't aware that Microsoft used AWS, seeing as they are very much enemies, as far as the cloud goes?

I've never seen this happen before?  I'm perplexed.  But, as I CAN download other zips from the github repository, I'm guessing that something really is different about this file?

SergeySlice, 2020-05-21:
All is good.

closed 2020-05-21 by SergeySlice