CloverHackyColor/CloverBootloader #170: CloverBootloader r5118

robaczek66, 2020-05-23:
When I update from r5107 do r5118 (Catalina 10.15.4) my graphic were destroy - menu, safari tabs other applications - a lot of blinks plus transparency lost, scrolling almost not work...
I tried to install r5117 and no change.
I go back to r5107 and everything is fine again.

I found that my graphic card  was not recognise correctly - in system preferences it says just -"Intel" when normally is - "Intel HD Graphics 4600"
The same for monitor

Is it mistake or I need to change some settings when update Clover? Any idea?

SergeySlice, 2020-05-25:
Set
~~~
	<key>Devices</key>
	<dict>
		<key>NoDefaultProperties</key>
		<false/>

~~~

closed 2020-06-01 by asava

robaczek66, 2020-08-11:
I use at the moment news version r5120... from first boot was some text appear very short like after it goes for normal Clover boot screen but with very very low resolution. It keeps that resolution with login and running newest Catalina 10.15.6 simply graphic was not recognised. I use your Advice and put manually into Plist:
<key>Devices</key>
	<dict>
		<key>NoDefaultProperties</key>
		<false/>
Bootloader screen not change same screen and resolution but just before login screen to MacOs resolution goes to normal and system works ok with correct graphic.
I can say like this but it was normal with all Clover and all Mac system few years back.
Any suggestion?
I attached my Plist
[image]

[image]
[image]
[image]

robaczek66, 2020-08-11:
My boot log:

0:100  0:100  MemLog inited, TSC freq: 3392143181
0:100  0:000  CPU was calibrated with ACPI PM Timer
0:100  0:000  
0:100  0:000  Now is 11.08.2020,  18:10:28 (GMT)
0:100  0:000  Starting Clover revision: 5120 (master, commit 33816ae90) on American Megatrends EFI
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED --conf=/Users/sergey/src/Clover/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SelfDevicePath=PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x4,0xFFFF,0x0)\HD(1,GPT,1C275948-AADB-19E7-8F39-9E8423BDF19A,0x800,0x64000) @CB82C118
0:100  0:000  SelfDirPath = \EFI\BOOT
0:100  0:000  Clover : Image base = 0xBD0B3000
0:100  0:000  SimpleTextEx Status=Success
0:100  0:000  === [ Get Smbios ] ==============================
0:100  0:000  Type 16 Index = 0
0:100  0:000  Total Memory Slots Count = 4
0:100  0:000  Type 17 Index = 0
0:100  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:100  0:000  SmbiosTable.Type17->Size = 2048MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 ChannelA-DIMM0
0:100  0:000  SmbiosTable.Type17->Vendor = 1302
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = 1600LLA Series    
0:100  0:000  Type 17 Index = 1
0:100  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:100  0:000  SmbiosTable.Type17->Size = 2048MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 1 ChannelA-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = 1302
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = 1600LLA Series    
0:100  0:000  Type 17 Index = 2
0:100  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:100  0:000  SmbiosTable.Type17->Size = 2048MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 2 ChannelB-DIMM0
0:100  0:000  SmbiosTable.Type17->Vendor = 1302
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = 1600LLA Series    
0:100  0:000  Type 17 Index = 3
0:100  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:100  0:000  SmbiosTable.Type17->Size = 2048MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 3 ChannelB-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = 1302
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = 1600LLA Series    
0:100  0:000  Boot status=0
0:100  0:000  Running on: 'Z87X-UD5H' with board 'Z87X-UD5H-CF'
0:100  0:000  === [ GetCPUProperties ] ========================
0:100  0:000  CPU Vendor = 756E6547 Model=306C3
0:100  0:000   The CPU supported SSE4.1
0:100  0:000  BrandString = Intel(R) Core(TM) i5-4670K CPU @ 3.40GHz
0:100  0:000   The CPU not supported turbo
0:100  0:000  MSR 0x35               40004
0:100  0:000  MSR 0xE2 before patch 1E000402
0:100  0:000  MSR 0xCE              00080838_F3012200
0:100  0:000  corrected FLEX_RATIO = E0000
0:100  0:000  FSBFrequency = 100 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:100  0:000  MaxDiv/MinDiv: 34.0/8
0:100  0:000  Turbo: 36/37/38/38
0:100  0:000  Features: 0x77DAFBBFBFEBFBFF
0:100  0:000  Threads: 4
0:100  0:000  Cores: 4
0:100  0:000  FSB: 100 MHz
0:100  0:000  CPU: 3400 MHz
0:100  0:000  TSC: 3400 MHz
0:100  0:000  PIS: 100 MHz
0:100  0:000  ExternalClock: 25 MHz
0:100  0:000  === [ GetDevices ] ==============================
0:100  0:000  GOP found at: PciRoot(0x0)/Pci(0x2,0x0)/AcpiAdr(0x80010100)
0:100  0:000  PCI (00|00:00.00) : 8086 0C00 class=060000
0:100  0:000  PCI (00|00:02.00) : 8086 0412 class=030000
0:100  0:000   - GOP: Provided by device
[... 333 lines cut ...]
4:226  0:000  ->Extra kext: EFI\CLOVER\kexts\Other\HibernationFixup.kext (v.1.2.3)
4:232  0:005  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC_SMMSensors.kext (v.1800)
4:239  0:006  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext.dSYM (v.1.2.4)
4:241  0:001  ->Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.3.2)
4:438  0:197  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC_LPCSensors.kext (v.1800)
4:647  0:208  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC_GPUSensors.kext (v.1800)
4:659  0:012  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC_CPUSensors.kext (v.1800)
4:668  0:008  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.2.7)
4:679  0:011  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC_ACPISensors.kext (v.1800)
4:693  0:014  ->Extra kext: EFI\CLOVER\kexts\Other\FakeSMC.kext (v.1800)
4:710  0:016  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Off
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10_normal
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15_normal
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6
4:710  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6_normal
4:711  0:000  SetStartupDiskVolume:
4:711  0:000    * Volume: 'Catalina'
4:711  0:000    * LoaderPath: ''
4:711  0:000    * DevPath: Catalina
4:711  0:000    * GUID = 1C275949-AADB-19E7-8F39-9E8423BDF19A
4:711  0:000    * efi-boot-device: <array><dict><key>IOMatch</key><dict><key>IOProviderClass</key><string>IOMedia</string><key>IOPropertyMatch</key><dict><key>UUID</key><string>1C275949-AADB-19E7-8F39-9E8423BDF19A</string></dict></dict></dict></array>
4:711  0:000  SetScreenResolution: 1024x768 - already set
4:711  0:000  Custom boot is disabled