CloverHackyColor/CloverBootloader #172: PluginType not working in 5118

CMMChris, 2020-05-25:
Hi,

I just noticed that PluginType injection stopped working in 5118.
[image]

Cheers,
Chris

SergeySlice, 2020-05-25:
May be the problem elsewhere?
[image]

CMMChris, 2020-05-25:
Hm, I will investigate and Report Back. Maybe ist's the latest macOS Beta.

CMMChris, 2020-05-25:
Yup, latest macOS beta is the culprit. It doesn't even work with SSDT-PLUG. Haven't been able to figure out the reason though. On my MacBook Pro X86PlatformPlugin does load just fine. Strange.

CMMChris, 2020-05-25:
Okay, issue solved, you can trash this ticket. Issue was invalid signature of X86PlatformPlugin. Somehow got corrupted during the last beta update. I restored the version of my CCC backup and it is working again.

closed 2020-05-27 by SergeySlice