CloverHackyColor/CloverBootloader #173: Hey,First of all thk u so much! ...and. -v is broken in r5118

APIECKAM, 2020-05-27:
(-v) is broken in r5118,& Thank you for your contribution to hackintosh  ; )

SergeySlice, 2020-05-28:
It works as usual.

mackonsti, 2020-05-30:
Hi @SergeySlice thanks again for your commitment to this project. But I am sorry to say that I agree with @APIECKAM because -v works indeed as a manually entered boot option BUT if you press SPACEBAR on the selected boot partition and move downwards to the Verbose (-v) Boot option, it does **not** boot like that.

Please try it by yourself and you will see it, it's been broken for a couple of releases now. I refer to the **quick options** for the selected boot partition, not the Clover Options window. Tried again now. Toggling the setting (depends on config _Attributes_ for boot) doesn't produce the expected results. Thanks!

cc: also @asava and @vectorsigma72 (thanks again guys)

asava, 2020-05-30:
There was an issue some time ago, but it's long fixed, and it works properly in latest releases.

I just tested again and it worked as expected. Including if you press spacebar and add it from there, then choose "boot with selected options".

Are you guys sure you don't have set nvram variable "boot-args" that contains -v? OpenCore may set boot-args if you are using it as well, which may explain issues.

mackonsti, 2020-05-30:
Hi @asava thanks for your reply. I just did that. I removed -v from my config.plist on Clover. Upon reboot I pressed F11 to clear NVRAM and booted. When reached the desktop in Terminal did `nvram -xp` and didn't see any `boot-args`. Rebooted. Note that in my config.plist there is no -v in **Arguments** key.

On Clover boot screen, pressed SPACE on my Catalina partition. The Verbose (-v) option was not checked as expected. Pressed space, the tick-mark appeared. Selected Return. Pressed ENTER to start booting from that volume-partition.

The boot logo appeared instead :( 

I understand about OpenCore NVRAM settings and I realised that they may not saved in the same memory or space etc. as Clover (from my experience on my other Hackintosh) but in this case, the config.plist Arguments `<key>` are respected but not the user-choice when pressing SPACE over a boot volume...

Your feedback is very welcome.

asava, 2020-05-30:
Oh, now I see why we have a different result.

The options you are ticking after pressing space are not saved to the main entry. They are taken into account only if after pressing space, you tag it, and choose "boot with selected options". 

Wasn't it always like this? (was there a different behavior in older revisions?)

mackonsti, 2020-05-30:
Thanks for confirming mate. As far as I can personally remember, no. The idea was (is?) to quickly select options for the highlighted volume/partition including Verbose (-v) boot, not necessarily verbose-action linked to booting with or without **injected** kexts.

So I did what you mentioned and yes, Verbose PLUS a selection of kext method (with or without injection) via SPACE bar did work as you say... But could I remember so wrong?

Anyway, as you know much more than me and can tap on the code, thanks for your feedback. It's up to you and the other great guys to see into this feature/functionality for Verbose via volume options (instead of Clover options)!

asava, 2020-05-30:
You don't need to necessarily select a kext method.
There is an option "Boot with selected options" in the submenu, which as it says, boot with the options selected below.

closed 2020-05-31 by asava

mackonsti, 2020-05-31:
Thanks @asava