CloverHackyColor/CloverBootloader #174: can't boot with 5117 or 5118

benedict6, 2020-05-28:
HI, I'm not sure if this is the same as issue $170. I can't boot in these 2 versions. I used verbose mode and saw an error about AppleKeyStore failed so tried nv_diasable = 1. It booted but gave me artifacts almost immediately.

In case this is important: I'm using Intel HD 4000 graphics. I also have the NVIDIA GeForce GTX 660 card installed but disabled it a long time ago because of the artifact problem in High Sierra. I never bothered to remove it from the motherboard though.

Also, this is probably unrelated so ignore if it is—I can't boot into Safe mode in at least versions 5116–5118. Don't know when this started though.

vectorsigma72, 2020-05-28:
> HI, I'm not sure if this is the same as issue $170. I can't boot in these 2 versions. I used verbose mode and saw an error about AppleKeyStore failed so tried nv_diasable = 1. It booted but gave me [...]

Hi, there's a change about a key in config, read [this ](https://www.insanelymac.com/forum/topic/329777-clover-problems-report-features-request/?do=findComment&comment=2723373) and see if solve the problem.

TonyC5, 2020-05-28:
I am experiencing the same problem with my Dell Latitude E6410 (Nvidia G3100m).  I use Clover (Legacy) with InjectNVidia=True.  I have reverted to Clover r5115 which works fine.

benedict6, 2020-05-29:
Yes, just tested and fixed here too. Thanks!

closed 2020-05-29 by benedict6