CloverHackyColor/CloverBootloader #175: Update 5108 to 5118 kernel panic

doadin, 2020-05-29:
Tried to update 5108 to 5118 no other changes and wouldn't boot.

asava, 2020-05-31:
Try to set:

	<key>Devices</key>
	<dict>
		<key>NoDefaultProperties</key>
		<false/>

TA2AGV, 2020-05-31:
I was stuck at 5108 because of black screen issue after boot menu with 5109.
@asava's suggestion solved it. Now I'm on 5118. Thanks!

asava, 2020-06-01:
duplicate of #170

doadin, 2020-06-01:
that actually didn't solve my problem

doadin, 2020-06-01:
thanks for closing the issue though

asava, 2020-06-01:
Not sure what you expected, as you provided no further information or response. I will reopen, but provide further detail on the panic you are having.

doadin, 2020-06-01:
well im just a hackintosh noob so not sure exactly what i should provide. A issue template would help with that. also just cause I didn't provide info instead of requesting more info and helping me know what info to provide, like "please pastebin logs found in xxxx" or something you just close the ticket as if its solved? ok i guess....

closed 2020-06-01 by doadin