CloverHackyColor/CloverBootloader #177: Errors when installing Catalina 10.15.4 and 10.15.5 with Clover Legacy

TonyC5, 2020-05-30:
When installing Catalina 10.15.4 or 10.15.5 from an installer USB stick with Clover Legacy boot, the Catalina installer has two errors as listed below.  These errors can be ignored and they do not happen when installing with Clover UEFI mode.

Error #1: After the first phase of the Catalina 10.15.4/5 installation, the installer terminates with 3 minutes remaining with the error "An error occurred while preparing the installation"

Error #2: After the second phase of the Catalina 10.15.4/5 installation, the installer terminates with "The bless tool was unable to set the current boot disk"

Both of these Catalina 10.15.4/5 installation errors can be ignored (although '/System/Volumes/Data/macOS Install Data' must be manually deleted after the installation) and these errors do not occur when Clover is booting in UEFI mode.  These errors do not occur in earlier versions of Catalina (e.g. 10.15.3).

asava, 2020-06-01:
I confirm this issue, and would like to add that in some cases #2 mentioned above may happen in UEFI environment as well.

asava, 2020-06-02:
This is just an assumption, but it might be related to EmuVariable. 
Could be that it doesn't happen where native NVRAM is used.

@TonyC5 the cases where you did not have it on UEFI, were all of them not using EmuVariable?

TonyC5, 2020-06-02:
@asava I didn't know that an EmuVariable driver was necessary (or even an option) when installing Clover legacy.  My clover legacy config (on three different legacy / bios systems) is as follows:

- Install Clover in the ESP
- Boot Sectors
-- Install boot0af in MBR
- Clover for BIOS (legacy) booting
-- Clover EFI 64-bits SATA
- BIOS Drivers, 64-bit
-- Recommended drivers
--- FSInject
-- File System drivers
--- ApfsDriverLoader
- Install RC scripts on target volume
- Themes
-- BGM
- Install Clover Preference Pane

After clicking "Continue", when prompted with "This Clover package is incompatible with this version of macOS," install anyway

asava, 2020-06-02:
EmuVariable is included in the firmware (boot file) on legacy clover, you can't choose if to have it or not, as native nvram is not an option.

Just trying to think what the issue is, and as I have a report on UEFI as well in a case where EmuVariable was used, I though it may be related.

You mentioned you have not seen it on UEFI environment, so I wondered if EmuVariable was used or not on your UEFI environment.

TonyC5, 2020-06-02:
@asava My only UEFI system uses the following Clover config.  With this Clover config on my UEFI system, I do not see the errors reported in this thread.

- Clover for UEFI booting only
- Install Clover in the ESP
- UEFI Drivers
-- Recommended drivers
--- AudioDxe
--- DataHubDxe
--- FSInject
--- SMCHelper
-- File System drivers
--- ApfsDriverLoader
--- VBoxHfs
-- Memory fix drivers
--- AptioMemoryFix
-- Additional drivers
--- NvmExpressDxe
- Install RC scripts on target volume
- Themes
-- BGM
- Install Clover Preference Pane

vectorsigma72, 2020-06-03:
> I confirm this issue, and would like to add that in some cases #2 mentioned above may happen in UEFI environment as well.

> When installing Catalina 10.15.4 or 10.15.5 from an installer USB stick with Clover Legacy boot, the Catalina installer has two errors as listed below. These errors can be ignored and they do not [...]

Hi, to code sign the Clover installer we need a paid certificate from Apple.... for a pkg that installs to protected paths (I don't know how many days it will take to be revoked among the account)... Clover.app instead can be authorized by a right click (as "from unidentified developer”) and doesn't care about 'macOS Install Data'. 

As asava said, the macOS installer write some variables to the nvram but if it is in a non working state these variables are not persistent across reboots. Must say that if you never run rc scripts nor the CloverDaemonNew this is not possible for a clean install (i.e. when the nvram.plist never was generated from a working OS....). Should be different when you update/upgrade with a working nvram, even emulated .. but always from within the OS. From a USB pen drive I wonder on who will write your nvram.plist... the answere is simple: no one.

A remedy would be to make Clover writing these variables when the installer is run in the early boot, but then you should know what. Not sure but for example these are needed:

1. IASInstallPhaseList     
2. IASCurrentInstallPhase  
3. etc.

closed 2020-07-26 by SergeySlice